hoglet67/RGBtoHDMI #209: Question on Apple IIc TTL support

timmcross, 2021-05-27:
I noticed Apple IIc TTL support in Beta30. Is that for the non-plus version? If so, that sounds awesome! What would the pinout be from the IIc to the RGB2HDMI?

IanSB, 2021-05-28:
@timmcross 

Apple IIc DB15	<->  RGBtoHDMI 12 way header
3 Sync		     8 Sync       -> Composite sync
10 GR		     4 Intensity    -> Enables the auto NTSC Artifact Color Decoding
11 SEROUT            9 Green3        -> Video Data
13 GND                 3 Ground       -> Ground

timmcross, 2021-05-29:
So I tried Beta 30 and the text mode was inversed.  HGR seemed to be correct.  I noticed Beta 31 was out.  Switched to that and wow.  Works perfectly.

This project has allowed me to end my search for retro monitors and not be forced to live with composite video.  I can't thank you and the other contributors enough!  Analog parts are coming next week for the IIgs.  Awesome work!

IanSB, 2021-05-29:
>I noticed Beta 31 was out. Switched to that and wow. Works perfectly.

Yes, there were a couple of bugs in Beta 30 that stopped IIc support from working.

Also Beta 31 added line by line switching of NTSC artifacts so that the four lines of text in mixed modes stay sharp white and don't get artifacts. If you use the analog interface on the composite video output you will get exactly the same quality but the four lines of text in mixed modes will get artifacts as there is no way to automatically detect that mode although full text screens will still be sharp white because the colour burst is switched off.

Note for PAL systems, the PAL encoder must be disabled to use the composite output and you may have to manually switch artifacting on and off using long presses of SW2.

>Analog parts are coming next week for the IIgs

The IIgs is unlikely to work properly with the analog interface as that can only discriminate 64 analog levels (2 bits / 4 levels each of RGB) but the IIgs can output 4096 levels (4 bits / 16 levels each of RGB). It should be possible to connect the 12 bits + sync directly to the IIgs motherboard instead to pick up the 4 bits each of RGB instead much like the Amiga and Atari ST but appropriate profiles will have to be created.

However there is a further complication with the IIgs because it uses two different pixel clocks, one of 14.31818Mhz to display the Apple II compatible modes and the other of 16.3636Mhz to display the new IIgs hires modes.
Because the pixel perfect sampling needs to know the pixel clock frequency in use, two separate profiles would be needed to capture the above modes and preferably some way to auto detect which clock is in use so that the software can auto switch between the two profiles.

There is a similar problem with the BBC micro which uses pixel clocks of 12 and 16 Mhz and that is auto detected by looking for a slight timing variation in the sync signals. I don't think that will work with the IIgs but there may be other signals on the motherboard which could be used to infer which frequency is in use but as I don't have a IIgs it is rather difficult to investigate this any further at the moment. Wost case you would have to manually switch between the profiles depending on which mode is in use but that is not ideal.

timmcross, 2021-05-30:
Ah, ok.  Thanks for the information.  I wasn't aware of that issue with the IIgs.  I have used it with a 15kHz VGA monitor splitting the sync signal as well as with a GBS8220.  I was thinking I could just send RGBS to the RGB2HDMI and it would work.

If you need someone to do testing, I have a ROM01 IIgs.

IanSB, 2021-05-30:
@timmcross 
>If you need someone to do testing, I have a ROM01 IIgs.

Thanks, that would be very useful.
I'm pretty confidant that I can create appropriate profiles for the two pixel clocks and if you are willing to solder up the 12 bits to your IIgs motherboard you can try them but as mentioned we really need to determine if there is a signal on the motherboard that can be used to infer which pixel clock is in use. The vertical sync input would normally be unused so that could be connected to such a signal to auto select the right profile.

If you are able to use an oscilloscope to look at some IIgs signals that would help a lot.

Unfortunately it doesn't look like there is a simple logic level available indicating the clock speed but one possible way might be to look at some of the signals coming out of the Mega II chip (which emulates the Apple II functionality). 
One of the signals out of the Mega II is SEROUT which is the mono video signal and that goes to TP154 and another is WINDOW which goes to TP155. If these are active in Apple II modes but 0V in IIgs modes they might be a possible signals to use.

NJRoadfan, 2021-06-06:
> Unfortunately it doesn't look like there is a simple logic level available indicating the clock speed but one possible way might be to look at some of the signals coming out of the Mega II chip (which emulates the Apple II functionality).
> One of the signals out of the Mega II is SEROUT which is the mono video signal and that goes to TP154 and another is WINDOW which goes to TP155. If these are active in Apple II modes but static in IIgs modes they might be a possible signals to use.

There is likely a way to determine if SHGR or classic Apple II modes are running on the IIgs. There was a clip-on VGC adapter available at one time that gave the ROM3 IIgs HDMI output: [http://tulip-house.ddo.jp/DIGITAL/gsdvi/index.html](http://tulip-house.ddo.jp/DIGITAL/gsdvi/index.html)

The ROM 01 wasn't compatible since it has a socketed VGC. Looks like the adapter compensated for the horizontal width difference by increasing the border size on the left and right sides in 14.318Mhz mode.

IanSB, 2021-06-06:
@NJRoadfan 

There is this comment on the web site you linked:

>Since this auto detection is not perfect, you can also use the adapter in manual mode

So it looks like it is inferring the state from some other signal or signals

IanSB, 2021-06-07:
@NJRoadfan 

Thinking about it, a standard Apple II video line is 912 clock cycles long. AIUI the IIGS VGC video clock is supposed to be 8/7 times the 14.318 Mhz clock which is 16.3636Mhz but if you multiply 912 by 8/7 you get 1042.285 so the line length and thus sync timing may vary slightly in the IIGS modes as it can't use a fractional line length. Maybe that's why it's not reliable as the detection is looking for a very slight sync timing change.

NJRoadfan, 2021-06-07:
@IanSB 

The SHGR modes on the IIgs were a bit fussy to adjust/capture using both the OSSC and directly with a RGB frame-grabber (Epiphan DVI2PCIe). I could get 640x200 mode dialed in pixel perfect, but always had a problem when the machine switched to 320x200 mode and had to re-adjust.

IanSB, 2021-06-07:
@NJRoadfan 

That might be an issue with the 640 pixel modes using a different clock phase to the 320 pixel modes. Something similar happens with an original IBM CGA card where the pixels for text and graphics modes are 180 degrees out of phase so sampling has to be set to 90 or 270 degrees as a compromise value. The Atari ST also has that sort of problem as the phase relationship between the sync pulse and the pixels shifts by 180 degrees randomly on power up so again a 90 or 270 sampling phase has to be used.

As I don't have any IIGS hardware this is going to be difficult to investigate any further.

timmcross, 2021-06-09:
I see what you are saying with the number of colors.  With the YUV CPLD, I can get a picture using the Apple II preset but the colors are off, even in the control panel.  Cycling through those colors doesn't show up correctly.  Red for example does not show up.  This is my only IIgs (ROM01) so let me know if you need me to check something.  Now that you have solved the IIc and Laser 128, this would be my last display issue to solve.

IanSB, 2021-06-09:
@timmcross 

Logging some info to start with would be useful.
Could you try the following:

Hook up your board to the IIGS composite output as above and display an Apple II compatible mode.
Then display the source summary page in the info menu and screencap it by pressing SW2 & SW3 simultaneously.
Switch to one of the IIGS super hires modes, exit and redisplay the source summary page and screencap it again.
(You must exit the menu and redisplay it as it might not get updated othewise)
Then use the save log and EDID option.
Finally post the two screencaps and log.txt from the SD card.

timmcross, 2021-06-09:
No problem.  For clarification, should I use what I already tested, connecting to the 15 pin video expansion or should I make a new cable and connect directly to the composite video out of the IIgs?  What would the cable pinout be to connect directly to the composite RCA jack to the analog board?

IanSB, 2021-06-09:
@timmcross 

It doesn't really matter as long as you are getting a locked image as I'm only interested in the sync timing.

timmcross, 2021-06-09:
Here are the files.  I used Copy II plus for the Apple II compatible mode, then I used the super high res test from the iigs dealer diags for the other mode.  Let me know if this helps.
[Captures.zip](https://github.com/hoglet67/RGBtoHDMI/files/6625762/Captures.zip)

IanSB, 2021-06-09:
@timmcross 

Thanks for those.

It looks like the sync timing is identical for 14.318Mhz and 16.363Mhz pixel clocks which means that sync timing must always come from the 14.318Mhz clock and there must be a phase discontinuity in the 16.363 Mhz pixel clock during horizontal sync as one line would be 1042.285 clock cycles long.

That in itself isn't an issue because the sample clock on RGBtoHDMI is run at 8x the pixel clock and is phase locked to the beginning of each line but the calculation of the actual sample clock frequency will have to be altered to take into account the additional fractional part of the line length.
 
How exactly have you connected the IIGS at the moment? (Which signals & which interface?)
If the analog interface did you fit U7 on the rear of the board?
Which profile and did you make any changes to it?

I should be able to give you a profile to test 16Mhz sampling with you current connection and if that works then 2 profiles could be created for 12bits per pixel at the two clock frequencies but in order to test that you would have to wire up the RGBtoHDMI directly to the motherboard to pick up the 12 bits directly from the hybrid DACs

timmcross, 2021-06-09:
I have the 6/8 bit RGBtoHDMI board with the Issue 3 Analog board.  I have R,G,B,S, and ground connected to the analog board from the DB15 connector on the IIgs.  I haven't fitted the 3rd chip on the rear of the analog board but I can add that.  I believe the Apple II profile was selected and I was trying different settings so I am not 100% sure where it stands.  I can reset the SD card and try it again if there is a certain profile you would like me to try.

IanSB, 2021-06-09:
@timmcross 

Unzip the attached file, overwrite the existing kernel.img file on the SD card with this one and copy the two AppleGS profiles into:
\Profiles\6-12_BIT_RGB_Analog on the SD card.

Then program the RGB CPLD and try selecting the two profiles as appropriate:

Apple IIGS HR for 14.3 Mhz Apple II compatible modes
Apple IIGS SHR for 16.3 Mhz Apple IIGS super hires modes

These are setup for 3 levels each of RGB so you will only get a vague approximation of the source image but the object at the moment is to get a clean image with no pixel jitters or flickering (Like your Apple II compatible screencap above).
At a minimum you will need to do an auto calibration but you may have to tweak the DAC voltage levels in the sampling menu as well as I have guessed the values.

Best test results will be achieved with peak white images only so if you have a mono SHR test image please use that.
 
As before if you can do two screencaps and a log save and post them that will let me analyse the results.

[GS_Test_1.zip](https://github.com/hoglet67/RGBtoHDMI/files/6626936/GS_Test_1.zip)

NJRoadfan, 2021-06-10:
@timmcross 

The GS/OS Finder desktop is a good test for 640 SHR mode. Load up a graphics program like Platinum Paint to test 320 SHR and create test patterns.

timmcross, 2021-06-10:
I copied the files over from the zip.  Reprogrammed the CPLD for RGB and switched to HR to test in Copy II plus.  Looks like it couldn't find sync.  I also started up GS/OS with SHR and it also couldn't find sync.  I tried the auto calibrate but no change.  Take a look at the screen caps and let me know what you think.  Also, if you have some guidance on adjusting the DAC levels, that would be great.
[Captures_061021.zip](https://github.com/hoglet67/RGBtoHDMI/files/6633088/Captures_061021.zip)

IanSB, 2021-06-10:
@timmcross 

First try adjusting the DAC-E Sync level to see if you can get the image to lock
If that doesn't work you can try sync on green by setting "Sync on G/V" to "On" then adjust DAC-F GV Sync to try for a locked image. (If you were using the Apple II profile originally, that was effectively setup for sync on green)

If you get a locked image you can try adjusting the Hi and Lo levels to threshold the incoming RGB video. The Lo value sets the voltage at which 50% RGB is displayed and the Hi level adjust the level at which 100% RGB is displayed. The Lo value should always be less than the Hi value for each colour. Note the red and blue channels share a setting but the green channel is separate.

Run the Auto calibration or manually adjust the Sampling Phase for noise free results.

Use the save configuration option after finishing

timmcross, 2021-06-10:
Messing with DAC-E didn't seem to help.  Turning on Sync on G/V and then adjusting DAC-F to even 0.00 got the sync to lock.  I tried increasing the voltage a little but wasn't sure how high to set it.  I did that for both HR and SHR modes.

Colors are still incorrect so I wasn't sure if I should fit the other IC on the back of the analog board now?  I took some more captures and saved the log again.  Auto calibrate couldn't get the picture perfectly stable and it seemed like the brightness or contrast was too high.  I didn't mess with those settings yet.  Let me know what you think.
[Captures_061021-2.zip](https://github.com/hoglet67/RGBtoHDMI/files/6634913/Captures_061021-2.zip)

IanSB, 2021-06-11:
@timmcross 
>Messing with DAC-E didn't seem to help

That probably means you haven't connected the sync input to the right pin or there is something wrong with the sync input on your analog board

>DAC-F to even 0.00 got the sync to lock

It looks like the IIGS outputs sync on green so that's why it works.

I can see there are some pixels that are purple which should be white so you probably need to tweak the G Hi and G Lo levels.
However the red and yellow vertical stripes across the screen look clean and stable enough to indicate that the sampling clock is working properly so we could proceed to the next stage:

This would involve soldering a ribbon cable from the 12 way IDC header on the main RGBtoHDMI board directly to pins on the motherboard, specifically the H2, H3 & H4 hybrid DACs which are the yellow resin SIL packages next to the VGC (they look like SIL resistor packs). 

With your existing board you would only get 9 bits per pixel instead of 12 but most images would look pretty close with only the least significant bit each of RGB missing. 

If that works OK you will need to get a 12 bit RGBtoHDMI board with 12 bit extender and wire that up instead to get all bits sampled correctly.

You would still have to manually change the profile until we found a suitable signal to control auto switching.

If you want to try the above, I'll create a couple of profiles for the direct connection and detail the wiring instructions.

timmcross, 2021-06-11:
Yeah, that sounds fine.  I will check the sync coming out of the 15-pin connector on the IIgs.  I wonder if that is no longer working.  I had it connected to a LM1881 and connected to a VGA monitor which worked fine for a while but then stopped.  As you said, it sounds like sync is also going out on green too.

I will order the 12 bit board and the extender if you could post the profiles and wiring instructions.  They will probably take a couple of weeks to come in.

NJRoadfan, 2021-06-11:
@timmcross 
Pin 3 is composite sync and Pin 12 is composite video on the DB-15 video port on the IIgs. Sync-on-Green should work as well.

IanSB, 2021-06-11:
@timmcross 

Here are the profiles for 9 bits per pixel (for your existing board) and 12 bits per pixel (for a new board):
[GS_Test2.zip](https://github.com/hoglet67/RGBtoHDMI/files/6640856/GS_Test2.zip)
(I also included the updated kernel.img file but it's the same as the last one)
Put the profiles in \Profiles\6-12_BIT_RGB and reboot

Looking at the IIGS schematic, the bit order of the RGB signals is unclear as the 12 bits coming out of the VGC are only labelled Vid0 - Vid11
https://mirrors.apple2.org.za/Apple%20II%20Documentation%20Project/Computers/Apple%20II/Apple%20IIGS/Schematics/Apple%20IIGS%20ROM%2001%20Schematics.pdf

It is likely that pins 2-5 of H2 are the blue bits, pins 2-5 of H3 are the green bits and pins 2-5 of H4 are the red bits but I'm not sure which end is the most significant bit (i.e. is it 2,3,4,5 or 5,4,3,2) so you will have to do some investigation and experimentation.

For 9 bit per pixel connection with your existing board:
Make up a 12 way ribbon cable with a 12 way IDC header on one end to plug into P2 on the RGBtoHDMI board.
(make sure to get pin 1 the right way around)
Solder the wires on the other end as follows, probably best to do on the underside of the motherboard as all signals are on through hole components:

DO NOT USE THIS WIRING FOR THE 12BPP BOARD AS THE WIRE NUMBERS ARE DIFFERENT!

Red Channel
Wires 7, 5, 1 to H4 pins 2, 3, 4 OR pins 5, 4, 3 depending on the bit order
 
Green Channel
Wires 9, 4, 10 to H3 pins 2, 3 ,4 OR pins 5, 4, 3 depending on the bit order
(Note this uses the Vsync input on RGBtoHDMI as one of the green bits in 9 bit mode)

Blue Channel
Wires 11, 6, 2 to H2 pins 2, 3, 4 OR pins 5, 4, 3 depending on the bit order

Sync
Wire 8 to H1 pin 3 (H1 appears to be green, not yellow)

GND
Wire 3 to H3 pin 7

+5v 
You can connect Wire 12 to +5v on the motherboard if you want to power the RGBtoHDMI from the motherboard instead of using a separate USB power supply but then you must make sure never to connect the USB power as that will feed +5v backwards into the motherboard. 

I suggest wiring up just one of the RGB channels plus the other signals first to find out which order is correct.
There is a video test program disk image here:
https://manilagear.com/downloads/
If you run the super hires 320 mode you can check the bit order for the channel you wired up which will look like this if you got it the right way around (except every pair of blocks will be the same colour due to missing bit)
https://manilagear.com/wp-content/uploads/2016/12/IMG_2364.jpg

Wiring up the 12 bit version is very similar but uses all 4 bits per RGB and the wire numbering is different as the extra 4 bits are wires 1 - 4 and the Vsync input is not used as one of the green bits.
To avoid confusion I will post 12 bit wiring details later if you manage to get the above working and want to try the full 12 bit board.

timmcross, 2021-06-12:
OK thanks for the info.  I have ordered the 12 bit boards.  I also checked the connection and found that the sync wire I was using wasn't making the connection.  I replaced it with a different wire and was able to turn off sync on green.  Not sure if you need me to send you more screen captures or logs at this point.

While I wait for the boards, I will look into wiring up to the DACs on the motherboard.  It doesn't look like it would be that hard.  It might take me a few days though as things are coming up.  Thanks again for the guidance!

IanSB, 2021-06-12:
@timmcross 

It would be useful to wire up your existing board as a 9 bit test so we can determine if it's going to work at all before wasting any time building up the 12 bit boards. (That would also determine the bit order so I can finalise the 12 bit wiring instructions - the first option in my RGB instructions above is the most likely bit order)

I've implemented manual switching between the 14Mhz and 16Mhz timings by pressing SW3 so even if we can't get auto switching working it will be easy to toggle between the two modes without having to call up menus.
I'll post the updated software in a few days once I've finished testing it.

BTW do you have an oscilloscope? That would be useful for looking at potential signals to use to determine if  a 14Mhz or 16Mhz mode is in use.

timmcross, 2021-06-13:
Sure, no problem.  It might take me a few days to find time to tear my GS apart but I will see what I can do.  And I recently purchased an oscilloscope so let me know what you need.

timmcross, 2021-06-16:
So I hooked up the red channel, sync, and gnd as you directed and copied over the profiles.  Here are some screen caps using the video test program you linked.  Let me know if it seems I am on the right track. 
[061521.zip](https://github.com/hoglet67/RGBtoHDMI/files/6659696/061521.zip)

IanSB, 2021-06-16:
@timmcross 

Yes that looks correct so I guess the first option in the RGB wiring bit order above is the right one.
In the SHR capture0.png image the red blocks are twice as tall as in the example photo linked above due to the missing bit in 9 bit sampling mode.

If you can hook up the green and blue channels we can next look at the sampling:
The SHR 640 pixel mode in capture2.png looks clean but the SHR 320 pixel mode in capture0.png & capture1.png looks ragged on the right hand side columns so there may be an issue with the sampling clock or sampling phase although the PPM error might indicate some issue with the timing calculation

Here is updated software which has manual switching support added and improved sample timing calculation:
Wipe your SD card and install the new files.

[AppleIIGS_test.zip](https://github.com/hoglet67/RGBtoHDMI/files/6662318/AppleIIGS_test.zip)

There are now only two profiles:
Apple IIGS (This is the 12 bit version)
Apple IIGS 9 bit (The one you should be using with your existing board)

To switch between Apple II compatible mode and SHR mode you can either press SW3 on the converter or select "Set1" or "Set2" from the "Timing Set" option in the main menu (Set1 is SHR & Set2 is Apple II)

As before, post some screencaps including the source summary

timmcross, 2021-06-16:
Looking great so far!  I had Set1 and Set2 backwards for a bit.  SW3 worked great.  Let me know if you need anything else.  The 12-bit boards I ordered from China have been made but not shipped yet...
[061621.zip](https://github.com/hoglet67/RGBtoHDMI/files/6664443/061621.zip)

IanSB, 2021-06-16:
>Let me know if you need anything else.

They look great and the timings in the source summary menu now look correct unlike the earlier SHR ones.

I think the only technical thing I'd like to see is a screencap of SHR 320 pixel mode with "busy" detail rather than the plain blocks to confirm that it is sampling pixels correctly and there is no issue like the one mentioned by njroadfan above. 
Generally some good looking 320 and 640 pixel graphics screencaps including the desktop would be nice as well to see what the SHR results look like.

@timmcross 

I've added the 12 bit Apple IIGS wiring info to the cables section:
https://github.com/hoglet67/RGBtoHDMI/wiki/Cables

Next thing is to investigate if there is a signal available on the motherboard that can be used to auto switch between Apple II and SHR modes. 

I've already added an auto switching option that switches between the two sets depending on the state of the VSYNC input. If that is low it selects set 1 and if it's high it selects set 2 and that might be all that is required if we can find a signal like that.

However it is also possible that the signal might vary in a different way:
e.g. a "busy" signal with lots of transitions per frame in Apple II mode and static in SHR mode or vice-versa which would require some additional software support to monitor.

There doesn't appear to be an explicit signal to switch between the modes as I think that is just a register bit in the VGC so it would have to be another signal that changes in some way when switching modes.

I think you will just have to probe the VGC and Mega II chip pins to see if anything repeatable happens when changing from an Apple II mode to an SHR mode.

Here are a few suggestions after looking at the schematic here:
https://www.applefritter.com/files/2021/03/18/Apple%20IIgs%20schematic.pdf

The Mega II chip ouputs  SERVIDEO on pin 21 (TP154) and WINDOW on pin 49 (TP155) and these might be busy in Apple II mode and static in HGR mode

The VGC has FLIP6 on pin 49 (TP24) and FLIP7 on pin 48 (TP25) which appear to change address bits to alter the memory map and I understand that the memory map is different in Apple II mode so those are other possibilities.

We really need someone who is familiar with the IIGS hardware to help with this.

timmcross, 2021-06-17:
OK, I was able to take some more screencaps tonight.  I will get out the oscilloscope tomorrow and see if I can take some readings.

Some other questions (forgive me if they are listed elsewhere in the Wiki):
- I noticed that some screens seem to start drawing the border near the top right of the screen.  The capture of the blue screen shows that pretty well.  Just curious what that is?  Would we not see it on a CRT monitor?
- Is there a good way to shut the pi down gracefully?  I feel removing the power abruptly may affect the SD card.  Looks like SW4 reboots the pi.  Could it be used to power down?
- Does the PI Zero handle this well enough or would it be better to use a PI3 or higher?
- I think I remember reading that systems with faster graphics would be too much for the RGBtoHDMI.  Could this be used with say a NES or other console?
[061621-2.zip](https://github.com/hoglet67/RGBtoHDMI/files/6667073/061621-2.zip)

IanSB, 2021-06-17:
@timmcross 
Thanks for the images, they look really good

>I noticed that some screens seem to start drawing the border near the top right of the screen. The capture of the blue screen shows that pretty well. Just curious what that is? Would we not see it on a CRT monitor?

Part of the video signal that is normally hidden off the top of the screen is being captured and that shows a few black video lines before the border starts. The capture values were just a best guess and have to be optimised on actual hardware.

You can adjust the capture area and position by adjusting the settings in the geometry menu as follows:

Change the Setup mode to "Set Min/Offset" (this will crop off most of the border)
Adjust the V and H offsets until the main part of the screen is visible and centred (The horizontal position will only be to the nearest 4 pixels)
Change the Setup mode to "Set Maximum"
Adjust the Max H Width and Max V Height for the maximum values that will capture the whole border without any black lines or columns around the edges

You need to do this twice, once in an SHR mode and once in an Apple II mode

When complete, save the configuration in the main menu and post the resulting file which will be in /Saved_Profiles on the SD card. I'll put that version in any new release.

IanSB, 2021-06-17:
@timmcross 

>Is there a good way to shut the pi down gracefully? I feel removing the power abruptly may affect the SD card. Looks like SW4 reboots the pi. Could it be used to power down?

The software is bare metal and there is no operating system. Once the software has booted, the SD card is only accessed when reading or writing a saved setting from the menus so there are no open files and nothing to 'safely shut down'. It essentially behaves like an 8 bit computer in this respect. 

>Does the PI Zero handle this well enough or would it be better to use a PI3 or higher?

The Pi2 and Pi3 are actually slower with this application because uncached memory access is slower with them and the extra cores don't help. It doesn't currently work properly on the Pi 4 as some of the integrated peripheral hardware has changed.

>I think I remember reading that systems with faster graphics would be too much for the RGBtoHDMI. Could this be used with say a NES or other console?

Probably not as it is limited to 12 bits per pixel and those 12 bits must be available digitally, not just an analog RGB voltages coming out of a chip. (It can handle up to 6 bits per pixel in analog mode)

timmcross, 2021-06-18:
@IanSB 
Thanks for the info.  I think I located a 5V connection on OS1 which is here next to the DACs.  I will use that when the 12-pin boards show up.

I played with the geometry settings and attached some captures and the profile.  Let me know if I scaled it correctly.  I wasn't sure how much border to leave.
[061721.zip](https://github.com/hoglet67/RGBtoHDMI/files/6674380/061721.zip)

IanSB, 2021-06-19:
@timmcross 
Thanks for the tweaked profile

>I think I located a 5V connection on OS1 which is here next to the DACs

I think there is +5v on Pin 10 of H1

IanSB, 2021-06-19:
@NJRoadfan 

Now it's basically working I've been looking at auto detection:
From the converter you linked above, it seems that most of the PLCC pins have been removed and it looks like only the following pins are in use:
62-68 & 1-6 (12 bits RGB + sync)
8    (VDD)
59  (VSS)
10  (14.3Mhz clock)

That doesn't leave many options for auto detection and the only two I can think of are:
1. Comparing the 14.3Mhz clock phase with the phase of the RGB bit transitions. These will always be in the same phase for the 14.3Mhz clock and vary in phase with the 16.3Mhz clock
2. Simply look for a border colour or video content in the extra 8 lines as Apple II modes are 192 lines and SHR modes 200 lines

I suspect it's using the first option as the problem with the second option is that some enhanced modes use the 14.3 Mhz clock as well as Apple II modes as this example from @timmcross shows:

[image]

This is a 14.3Mhz Apple II style text mode but has colour and a border.

It might be possible to implement a phase comparator in the CPLD to get the first option to work if no other simpler signal is found.

NJRoadfan, 2021-06-19:
@IanSB The color text is a built in function of the Apple IIgs. You can assign any color to the text, text background, and overscan border in the control panel or a programmer can change it via a softswitch like Dagen did. The multi-color border is a programming trick that switches the border color register at a very high speed. The demo scene used it a bit to put graphics and animation in the overscan area (see FTA's Split demo in the XMas demo).

IanSB, 2021-06-19:
@NJRoadfan 

Thanks for the info, it looks like the text may still originate in the Mega II chip and then get colourised in the VGC so monitoring the ouput of the Mega II chip might still work but it depends on what that chip outputs when an SHR mode is being used. 
If it's output is blank during SHR modes then it could be used but if it just outputs garbage then it wouldn't be suitable.

NJRoadfan, 2021-06-19:
@IanSB The text screen comes from the Mega II. It should be noted that the composite output on the IIgs has a color killer that activates whenever text mode is on (looks like its controlled by VGC pin 7 mono/color). This was done for readability reasons as the color fringing was pretty intense.

IanSB, 2021-06-19:
@timmcross 
> I wasn't sure how much border to leave

Ideally the profile should capture all the border because a variable amount of the border up to the maximum will be used in integer scaling mode so perhaps you can try again adjusting the max settings to capture the entire border.

For the Max H Width, increase the value until the next adjustment adds no further border to one or the other side of the image then back off by one or two clicks.
Similarly for the Max V Height increase the value until you see a black line at the top or bottom and then back off again by one or two clicks.

This should be repeated for an SHR mode like the desktop and an Apple II mode like the above text screen.

After that, if you want to reduce the border on your particular screen which is using interpolated scaling you can use the crop border option in the Preferences menu but the default profile should have as much border as possible

IanSB, 2021-06-19:
@NJRoadfan 
>the IIgs has a color killer that activates whenever text mode is on

That's interesting as that could be used in conjunction with the above 192/200 line test to provide another possible way to determine the mode if the Mega II output idea doesn't work.
i.e. 
colour killer on = 14.3 Mhz Apple II text mode (even with colour overlay)
colour killer off = either 16.3Mhz SHR or 14.3Mhz Apple II compatible graphics (which will only have 192 line video and no border)

timmcross, 2021-06-27:
Thanks again for all the information.  I have received the 12-bit boards today and quickly put them together.  Here are some screen captures and the logs.  I think it looks correct.  Let me know if you want to see anything else.
[062621.zip](https://github.com/hoglet67/RGBtoHDMI/files/6720963/062621.zip)

IanSB, 2021-06-27:
@timmcross 
>Here are some screen captures and the logs. I think it looks correct

Yes they look correct

I received some feedback from Adrian Black (Adrian's Digital Basement on Youtube) about auto detection of the SHR clock and it looks like a couple of the lines I suggested above (flip 6 & flip 7) might be usable so I'm implementing a test version using one of those signals at that at the moment.

To test this you will need to connect flip 6 to the vsync input on the 12 bit board

flip 6 is on the following locations:
Pin 49 of the VGC
Pin 2 of UF3 (74F86)
TP 24 (may be on underside of board?)

IanSB, 2021-06-27:
@timmcross 

Here is a test version of the new auto select code:
[AppleIIGS_Auto_Test_Alpha1.zip](https://github.com/hoglet67/RGBtoHDMI/files/6722220/AppleIIGS_Auto_Test_Alpha1.zip)

As before, wipe your SD card and copy the new files on to it
You will be prompted to update your CPLD to RGB v93 on first powerup (This contains changes for the IIGS auto detection)

As I'm not sure if the auto detect is 100% reliable (or if it will work at all!) you can still manual override by pressing SW3 as before
The LED D2 on the RGBtoHDMI board indicates which mode it is in:
D2 Off = Apple II compatible 14.3 Mhz
D2 On = SHR 16.3 Mhz

I've updated the Cables page in the wiki with the full 12 bit connection info for the IIGS:
https://github.com/hoglet67/RGBtoHDMI/wiki/Cables

timmcross, 2021-06-28:
Wow, this definitely worked as described.  Great work coding this without a system to test on.  I used the test point connected to pin 49 on the VGC (I assume this is TP24 -- used my multimeter to find it) and I connected the 5v line to H1, pin 10.  D2 turned on and off as the mode changed.  I see no issues here so far!  I included a screenshot of my cable connections.  I will test with other apps this week.  Now I just need to find a way to mount it...  Maybe on or hanging off the PS with the HDMI running out the back.  I may get a HDMI panel mount connector to create a HDMI port on the back.

I auto calibrated the sampling and saved the profile.  Let me know if you would like any other info.
[062721.zip](https://github.com/hoglet67/RGBtoHDMI/files/6722954/062721.zip)

I can't thank you enough for this adventure.  Glad I was able to help test this configuration for you.

IanSB, 2021-06-29:
@timmcross 

Thanks for the update.
I suppose an internal clip on PCB could be designed similar to the DVI interface linked above by @NJRoadfan for systems with a soldered VGC.
If you want to fit the converter internally, you can switch the menu control to single button mode by editing the default.txt file in /Profiles (Set single_button=1). After that, the menus can be driven by short and long presses of SW1 so you only need to have a single button available outside the case.

IanSB, 2021-06-30:
@timmcross 

There is a new Beta 33 release on my fork which has all the recent changes consolidated. This will likely be a release candidate for the next stable version unless bugs are found so could you check it is OK for you (Some minor changes since the version above):
https://github.com/IanSB/RGBtoHDMI/releases

timmcross, 2021-06-30:
I copied beta33 over and booted up with that.  Everything behaved as before.  After seeing posts about the sparkling on the Tandy, I think I see something similar in SHR.  I see black and white dots popping up every so often.  It was worse before I auto calibrated it.  I also have the PI powered from the IIgs.  I could try to go back to it's own power supply.  It doesn't bother me too much but I haven't used it for that long.

I tried to capture it on video.  Let me know if you can see it.
https://youtu.be/X2Lv53bk-bU

IanSB, 2021-06-30:
@timmcross 

Try going into the settings menu and increasing the overclock core by 10 or 20 or even more but high values may require a heatsink on the Pi zero.
Save the config before exiting the menus
You can also try manually adjusting the sampling phase but the auto option looks like it is already picking a sensible value.
The sampling phase needs to be set separately in Apple II mode as well
If you get a good result, post the saved profile (will be in /Saved_Profiles)

Changing to external power is very unlikely to change anything as this connection is all digital. The issues with the coco are due to marginal analog signals. Any problem here is due to the Pi not being quite fast enough.
The current overclock core setting was used at 16Mhz on the Atari ST but you may need a slightly higher value as the clock is 16.3 Mhz

timmcross, 2021-07-08:
Sorry for the delay.  It looks like with the extender board, there is very little room for a heatsink on the PI.  I tried increasing the PI to 10 and 20.  At 20, it seems that the dots are still there.  I noticed the PI was up to about 45 degrees.  At what temperature would it be too hot?  I think the PI zero and 3 both share a similar processor so I will try it with a PI3 and see if I notice any difference.

For me, I can ignore these dots.  Let me know if there is anything else you would like me to try.

IanSB, 2021-07-08:
@timmcross 
>At what temperature would it be too hot? 

Mine run at 55 degrees so 45 is no problem. Most Pi zeros will overclock core up to 140+ without a heatsink although some won't.

The system doesn't run very well on other Pi models so don't bother, it won't work.

Anyway, please try the latest beta 36
https://github.com/IanSB/RGBtoHDMI/releases
You will have to reprogram the CPLD again with the version in this release.
Try running an auto calibration (menu option or long press of SW3)
If that doesn't help try increasing the multiplier in the sampling menu and run the calibration again (or manually adjust the sampling phase in the sampling menu).

After trying auto calibrations at various multiplier settings, go to the info menu and use the save log function and post the log.txt file from the sd card.

Maybe record a video of the calibration running as well?

Are the dots all across the screen evenly or just in vertical bands? (i.e. clean areas and noisy areas). (That would point to an incorrect geometry setting)

aotta, 2021-07-08:
i'm going to cabling my IIgs, but not sure about wiring... from wich side i'd start counting pins in the resistor arrays? can someone post a pic of soldered cable?
i know, i can read schematics and use the multimeter but.... i'm lazy and it's very hot this summer's nights! ;)

timmcross, 2021-07-08:
The square pin is pin 1.  Here is a pic of the cable I soldered on.  The red stripe is pin one on the cable.
[image]

aotta, 2021-07-08:
Thank you very much @timmcross , so you used (was my first doubt) the bottom side of the board for soldering... but i can't see which point you used for vsync (wire 14)... can you tell me?

aotta, 2021-07-08:
@timmcross sorry for previous message, i noticed now you used only 15 wires! (i usually connect the 5v for powering rgbtohdmi).
So, all connection are clear now for me (and didn't look under the board to see the square pin in arrays before, sorry!)
Thank you again!

timmcross, 2021-07-08:
Actually, pin 14 needs to go to the VGC so I used the TP at the top of the picture, under R198.  I believe I have pin 16 connected to pin 10 on H1 for +5v power.

aotta, 2021-07-08:
it works for me with beta 36, and auto calibrating works too.
I've still some issue since border color sometimes change and it seems lost the some blue signal, but it's sure a poor cabling, tomorrow i'll resold them and test again. Thanks @timmcross  for support!

[17 more replies]

closed 2023-12-27 by timmcross