hoglet67/RGBtoHDMI #211: AKAI PG1000 CRT issue

jczmok-iag, 2021-06-12:
Hi, i've got an old Akai PG1000 here where i try to get the rgbtohdmi running with.
I shortly see the AKAI logo "partly", it does not seem be in sync.

https://www.acc.umu.se/~polarn-p/irr/99musik/Akai_dokument/Akai_Digital_Matrix.pdf
-> MZ1000 tech data (is the original monitor).

as it is DB9 i would assume it is CGA, but i would to verify:
- cable is correct => does the numbering on the MALE adapter on the solder side start with 1 on right top ? It's a bit unclear in the documentation what the CGA DB9 wiring should be.
- - need setting up a profile on the rgbtohdmi for cga/mga adapter type. has this already been done ?

Thanks for helping me out.

jczmok-iag, 2021-06-12:
This looks like when i am using CGA mode ( AKAI logo shortly appears, but does not look right)

[image]
[image]
[image]

I've used this wiring:
https://github.com/hoglet67/RGBtoHDMI/wiki/Cables#ibm-pc-mda-cga-ega-2-4-or-6-bit-ttl--olivetti-prodest-pc1-4-bit-rgbi-ttl--commodore-128-80-column-4-bit-rgbi-ttl

however what seems odd to me, when i am looking up CGA wiring, there is some discrepancy:
http://brassicgamer.blogspot.com/2017/06/cga-ega-serial-modems-difference-how-to.html
[image]

Please avise...

IanSB, 2021-06-13:
>however what seems odd to me, when i am looking up CGA wiring, there is some discrepancy:

The wiring for RGBtoHDMI supports EGA as well as CGA. In CGA mode, Pin 6 is intensity and pins 2 & 7 are ignored. In EGA mode Pin 6 is secondary green, pin 2 is secondary red and pin 7 is secondary blue

>I shortly see the AKAI logo "partly", it does not seem be in sync.

The pdf documentation says that the monitor sync input is 5v digital TTL level negative going composite and the RGB inputs are 0.8v level analog.
In the CGA standard, the sync levels are 5v digital TTL level positive going separate H & V and the  RGB signals are also 5v digital TTL levels like the sync so the output of the AKAI isn't CGA at all, it just uses a similar connector and that is why it doesn't work because RGBtoHDMI is expecting approx 5v RGB signals not 0.8V analog signals and separate syncs instead of composite sync.

To get the image to lock you could go into the geometry menu and change the Sync Type to composite but that won't help with the wrong level RGB values (Save the config before exiting the menu)

You might be able to get it working using the analog addon board but it would depend on how many levels of analog video were being output by the AKAI.

Are there lots of shades of colour/grey on the output or is it just peak RGB / White?

jczmok-iag, 2021-06-13:
Originally (by looking at -> https://www.acc.umu.se/~polarn-p/irr/99musik/Akai_dokument/Akai_Digital_Matrix.pdf ) I only see b/w. Okay, then time to get the analog board ordered and testing then. I keep you updated!

jczmok-iag, 2021-07-07:
Okay, analog board is mainly assembled. Any wiring idea ? (and see my other question about D1)

IanSB, 2021-07-07:
@jczmok-iag 
>. Any wiring idea ?

CGA pin 1 to 6 way Pin 1 (GND)
CGA pin 8 to 6 way Pin 2 (Composite sync) - Possibly CGA Pin 9 to 6 way pin 2 instead if composite on Vsync instead of Hsync
CGA pin 5 ro 6 way Pin 3 (BLUE)
CGA pin 4 to 6 way Pin 4 (GREEN)
CGA pin 3 to 6 way PIN 5 (RED)

Pin 6 of 6 way (5v in) not connected, power the Pi zero with a 5v usb adapter

After connecting: copy this profile into the /Profiles/6-12_BIT_RGB_Analog on the SD card, reboot the Pi zero and select it with the profile line:
[AKAI PG1000.txt](https://github.com/hoglet67/RGBtoHDMI/files/6779740/AKAI.PG1000.txt)
This profile is set up for CGA timing with 5v composite sync and 75R terminated .8v RGB signals as described in the Akai manual

You may need to tweak some of the RGB Hi & Lo levels in the sampling menu but give it a try and post some screencaps and the System summary menu

jczmok-iag, 2021-07-09:
Finally found a quiet minute to finish the wiring (according to your setup): IT'S WORKING.
Startup looks a bit washy, but once it's started, Display works. You can now officially add this setup to "supported" list.

[image]
[image]
[image]
[image]

IanSB, 2021-07-09:
@jczmok-iag 

Did you run an auto calibration to set the optimum sampling point?
If so what sampling phase number was chosen?
(I'll make that the default)

jczmok-iag, 2021-07-10:
Well as the Message was blinking , nevertheless i did try a auto calibration but nothing changed. seems to be good so far.

IanSB, 2021-08-04:
I've added the profile to the RGB Analog set but it's called 'Analog_CGA' as it might be useful on other systems with similar analog outputs with CGA timing.

closed 2021-08-04 by IanSB