hoglet67/RGBtoHDMI #213: Are the kits available for sale? Need to convert 3-bit RGB into HDMI

alex3331, 2021-06-22:
Are the kits available for sale? Need to convert 3-bit RGB into HDMI

IanSB, 2021-06-26:
They will be shortly.
I'll update when available.

alex3331, 2021-06-26:
Please do. I am interested

jczmok-iag, 2021-07-03:
If that helps, i have some RGB2HDMI pcb's and and analogue pcb's and most parts for those. If there's interest...

alex3331, 2021-07-03:
Will you have assembled RGB2HDMI boards? How much will this cost?

jczmok-iag, 2021-07-03:
Only parts. I ordered bare PCB and 99% parts from the sources. Approx (unassembled) it is around 25 EUR each with most parts (shottky diode missing)

alex3331, 2021-07-03:
I need just a PCB., on which xc9572 is mounted
I have parts.

jczmok-iag, 2021-07-04:
i'd say 30 EUR but i need to have most parts mounted in order to test CLPD function. contact me off list jan (at) czmok (dot) de

hougaard, 2021-07-18:
Any updates on the availability of kits (or even better, complete units), I've paused several projects in anticipation of getting my hands on a few boards.

Thanks!

jczmok-iag, 2021-07-18:
Am 2021-07-18 06:53, schrieb Erik Hougaard:
> Any updates on the availability of kits (or even better, complete
> units), I've paused several projects in anticipation of getting my
> hands on a few boards.
> 

Hi, i have about

12 unpopulated analog board here
8 rgb2hdmi boards (but waiting for cpld's)

i can provide fully assembled & tested rgb2hdmi boards
or a kit with all parts, but cpld soldered.

let me know what you need and i can provide you totals (and include your 
location) regarding
shipping costs.

when those things are sold, they are sold. i am not reordering. just 
needed to fill the minimum
shipment requirements.

--jan

hougaard, 2021-07-18:
That sounds amazing!!

I'm in North Vancouver, British Columbia, Canada.

I really need two fully assembled boards. One for doing CGA (out of an IBM Color Graphics Adapter) and one for "others". If I understand rgb2hdmi correctly, the CGA can be done without the analog board? 

So two boards, one with analog?

Cheers

hougaard, 2021-07-18:
Let's move this out of Hoglet's issue thread, email me on erik at hougaard dot com - thx!

closed 2021-08-04 by IanSB