hoglet67/RGBtoHDMI #215: Question on compatible CPLD

timmcross, 2021-06-26:
Seeing that the two CPLDs mentioned in the wiki are out of stock:

XC9572XL-10VQG44I
XC9572XL-7VQG44C

there is another which is available:  XC9572XL-7VQ44I‎
https://www.mouser.com/ProductDetail/?qs=rrS6PyfT74c4YzeFShmWVg%3D%3D

Can this be used in place of the other two?

Can any XC9572XL with 44 pins be used?

IanSB, 2021-06-26:
>Can any XC9572XL with 44 pins be used?

It needs VQ in the part number as that means Quad Flat Pack (VQFP)
The I and C versions just have different temperature ranges  (Industrial and Commercial?)

>there is another which is available: XC9572XL-7VQ44I‎

That should be fine, the only difference is that it isn't RoHS lead free (Missing the 'G' in the part number)

timmcross, 2021-06-27:
Great, I will order some more CPLDs today to complete the other 10 boards I have.

closed 2021-06-27 by timmcross