hoglet67/RGBtoHDMI #216: Atari STe color swapped

pixelk0, 2021-06-30:
Hello and first, thank you. I already posted about some trial and errors on the ST Thread from [ c0pperdragon /
Amiga-Digital-Video ](https://github.com/c0pperdragon/Amiga-Digital-Video/issues/6).

I now have a perfectly stable image (needed to run the auto-calibration) but I have spotted a problem in the colour handling.

First it may be an error while soldering the 12 bits inputs, but the consistency in my error seems strange, it's as if some bits are swapped for all 3 RGB channels. A colour with a higher value is darker than the precedent, but it only happens in one in two colour.

Maybe it will be clearer with a few screenshots (I noticied it first trying some demos, then deluxe paint and degas elite)

[image]
[image]
[image]
[image]

https://user-images.githubusercontent.com/5732483/123918263-fd630400-d983-11eb-966e-99be3ca8676b.MP4

https://user-images.githubusercontent.com/5732483/123918289-048a1200-d984-11eb-95e5-33b36be0432c.MP4

https://user-images.githubusercontent.com/5732483/123918302-081d9900-d984-11eb-8977-bf07edba11cd.MP4

Is my diagnostic correct (I swapped bits when soldering) ? 
Have anybody encountered the same problem ?
Can I just modify a palette or something else to correct this in software ?

IanSB, 2021-07-01:
That looks like you have the 4 bits of each colour rotated by 1
i.e

Atari R0 to Pi R1 
Atari R1 to Pi R2
Atari R2 to Pi R3 
Atari R3 to Pi R0
 
Same for the other colours I think

pixelk0, 2021-07-01:
OK, thanks, I'll redo my soldering (and my photos for future installations)

IanSB, 2021-07-01:
@pixelk0 
Note I have changed the above table as I think I got the rotation in the wrong direction

pixelk0, 2021-07-01:
I mis-identified the pins on the STe's motherboard, and forgot that the bitness is reversed, here is the definitive pinout for the STe to the RGB2HDMI :

[image]

This issue can be closed, thank you so much IanSB !

closed 2021-07-01 by pixelk0

luciust, 2022-06-15:
@pixelk0 @IanSB 
Thanks! I can confirm that it fixes my issues too from: https://github.com/IanSB/RGBtoHDMI/issues