hoglet67/RGBtoHDMI #217: NTSC artifacting for the CoCo 2

qbancoffee, 2021-06-30:
I was testing Donut Dilemma on for the CoCo2 and I can't seem to get the colors to match up with what the game looks like on 
the online CoCo emulator.
https://www.6809.org.uk/xroar/online/ 

Here are some captures from the RGB to HDMI board compared to screen caps on my computer
<H1>RGB to HDMI</H1>
[image]
[image]
[image]
[image]
[image]
[image]
<H1>Computer screen caps</H1>
[image]
[image]
[image]
[image]

IanSB, 2021-07-01:
Those are actual colours, not artifact ones and the standard RGBtoHDMI palette generates them mathematically from the original YUV values in the 6847 datasheet so the XRoar values may have some artistic licence. However I also included a palette to match XRoar so go into the palette menu and change the palette to "Dragon CoCo Emulators" and see how that looks.

qbancoffee, 2021-07-01:
I think your right about the artistic license, I've been looking at the authors website and although the images he has on display are for a coco 3, they are closer to what is being output by the pi.
https://nickmarentes.com/ProjectArchive/donutcc.html

qbancoffee, 2021-07-01:
> However I also included a palette to match XRoar

Doesn't seem to match up with Xroar with any of the palettes but I'll take the screenshot and ask on the coco discord server.

IanSB, 2021-07-01:
>Doesn't seem to match up with Xroar with any of the palettes but I'll take the screenshot and ask on the coco discord server.

I think the emulator palette values came from Xroar but it was a couple of years ago so they may have been changed in xroar since then

qbancoffee, 2021-07-01:
Tomorrow I'll reconnect the RF can and hook it up to my tv to see if the colors are close.
Nick Marentes has some PAL images that match what I'm seeing.

https://nickmarentes.com/ProjectArchive/pipes.html

If you scroll down he explains how you'll only see this correctly on a PAL tv.

IanSB, 2021-07-01:
>If you scroll down he explains how you'll only see this correctly on a PAL tv

He's using PAL artifacting which uses different colours on adjacent lines to get the PAL decoder to average to a new colour.
PAL artifacting isn't supported yet but there is a reserved option for that in one of the menus.
I think some Commodore 64 demos use that as well.

qbancoffee, 2021-07-01:
I connected the rf can to my coco2 and loaded donut dilemma and except for the background pattern, it looks very similar but still a little off. Do you think this is close enough?

[image]
[image]
[image]

qbancoffee, 2021-07-02:
> PAL artifacting isn't supported yet

Oh imissed that.

IanSB, 2021-07-02:
Other than the red/blue artifacting I think my colours are a better match for the off screen, look at the orange donuts, they are almost red on Xroar.

That program isn'r really useful for testing the standard colours though. 
Are there any video test programs that run through all screens displaying all colours as that would be more useful.
In the meantime, an off screen photo of the NTSC RF output displaying the for/next loop colour blocks might be useful

qbancoffee, 2021-07-02:
> Are there any video test programs that run through all screens displaying all colours as that would be more useful.

I've been looking around but I don't see one.

> I think my colours are a better match for the off screen

Yes, I should have said that your implementation is a better match.

> In the meantime, an off screen photo of the NTSC RF output displaying the for/next loop colour blocks might be useful

o.k. ill hook it up and check

qbancoffee, 2021-07-02:
It look horrible but....

[image]

qbancoffee, 2021-07-02:
I can't really get a good picture of it with my phone. The yellow looks green in the picture but not on the screen

qbancoffee, 2021-07-02:
[image]

IanSB, 2021-07-02:
The cyan looks very blue as well is that the same in person?

IanSB, 2021-07-02:
Here's a PAL off screen:
[image]

qbancoffee, 2021-07-02:
> The cyan looks very blue as well is that the same in person?

It's pretty close in person. Here is a better picture and anything that is in the center of the pic is lighter than in person.
[image]

qbancoffee, 2021-07-02:
So a couple of days ago I saw a TV in a trash pile and I just drove over and trash picked it in the rain.....
To my delight and my wifes' dismay it works.

Here's a picture.

[image]

qbancoffee, 2021-07-02:
This video better shows what the colors actually look like.
https://youtu.be/qtcAOyhdAxs

IanSB, 2021-07-02:
I've tweaked the palette generation algorithm and it's now very close to the PAL output:
[image]
[image]

The NTSC output is a bit indeterminate as it is affected by phase shifts. You might be able to get it to match the PAL output by adjusting the tint control on the TV.

qbancoffee, 2021-07-02:
> You might be able to get it to match the PAL output by adjusting the tint control on the TV.

Yes, and although not all sets were made the same, I made sure to set the color/tint to default before testing.
I have a broken coco composite board that I'll try and repair to see if I can get better results.

IanSB, 2021-07-02:
You can also adjust the tint setting in the palette menu to get the palette to match the tv.

qbancoffee, 2021-07-09:
I posted this under the wrong issue earlier

Here are two videos where I compare the colors on my dell monitor and my television set.

https://youtu.be/wPlHFHk4y48
https://youtu.be/dPttAgc3Bog

I also noticed that the aspect ratio isn't quite 4:3, it seems longer on horizontal.(I measured it and it's 240mm X 318 mm so it is 4:3)

As far as the palette goes, I could sit in from of my television set and come up with numbers
that are closer to what I am seeing. If you want me to, where in the code should I look?

qbancoffee, 2021-07-09:
I posted the links to the videos I made on CoCo Discord and asked to see which is closer to what it should look like.
I guess the consensus is that the old TV is closest. Here is are some of the responses.

_"Except for the white comparison (white isn't actually supposed to be white, it's "buff"), your observations seem pretty right."_

_" Looking at your two videos above. The Yellow, Blue and the Red (CLS3 and CLS4) are definitely a little off from what "normal" TV's did back in the day (the yellow might be acceptable; while it's an orangish yellow, it is still a yellow compared to the regular orange (CLS8). Your NTSC TV looks exactly as I remember it being on older TV's."_

_"For the orange text character set - the colors should be really dark red (darker than CLS 4) on a yellowish/orange background (which is not the same as regular yellow (CLS2) or regular orange (CLS8). To see the dark red more clearly, you can do this from the OK prompt: POKE 359,57:POKE 65314,8. Then hit SHIFT-0 for lowercase and type a couple lines of text."_

IanSB, 2021-07-09:
>If you want me to, where in the code should I look?

The relevant code is is osd.c, look for:
case PALETTE_ATOM_MKI_FULL:
(The name dates back to before dragon/coco support etc)
However it is generated algorithmically from the YUV voltage values but doesn't take into account other things like gamma correction.

What you could try is to adjust the tint and contrast settings in the palette menu for each colour individually ignoring the others and post the list of those settings that make each colour match the TV.
i.e. concentrate on RED, and adjust tint and contrast until RED looks closest to the TV (ignoring the other colours)
Then repeat for each colour in turn so you end up with a list of those settings for each colour.
I'll apply those adjustment values to the palette for each colour and the resulting palette should then match the TV

qbancoffee, 2021-07-09:
> What you could try is to adjust the tint and contrast settings in the palette menu for each colour individually ignoring the others and post the list of those settings that make each colour match the TV.

o.k. will do.

qbancoffee, 2021-07-10:
o.k. , that was very hard.....
I could match the colors exactly by varying just the contrast and tint but they did improve.

Using the CoCo-emu palette, these are the colors that needed changing followed by the contrast and tint

| # | Color | Contrast | Tint |
| --------------- | --------------- | --------------- | --------------- |
| 2 | yellow | 89 | -33 |
| 3 | blue | 149 | 20 |
| 4 | red | 124 | -6 |
| 6 | cyan | 88 | -54 |

qbancoffee, 2021-07-10:
Would it help If I gave you RGB values that you could use to apply corrections?

IanSB, 2021-07-10:
@qbancoffee 
>Would it help If I gave you RGB values that you could use to apply corrections?

OK, I was trying to keep it algorithmically but I'll try using an RGB override on some colours

IanSB, 2021-07-10:
@qbancoffee 

What is the best default for the NTSC phase? (0 or 180)
(I suppose it depends on how many games use each one as their default)

qbancoffee, 2021-07-10:
o.k.
 I asked my wife to help me out because she is much better at matching colors than I am.
We used Kolour paint to match the colors visually and then I just copied the the values.
You don't need Hue and saturation but they were there so I added them here.
Also, I think the green already looks good.

| CLS# | Color | R | G | B | Hue | Saturation |
| ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- |
|1|green||||||
|2|yellow|216|254|0|69|100|
|3|blue|1|74|253|223|100|
|4|red|254|0|0|0|100|
|5|buff|205|210|224|224|8|
|6|cyan|57|180|216|194|74|
|7|magenta|241|67|194|316|72|
|8|orange|248|105|19|22|92|

qbancoffee, 2021-07-10:
> What is the best default for the NTSC phase? (0 or 180)

So far 0 matches the games I'm using.

closed 2021-08-04 by IanSB