hoglet67/RGBtoHDMI #221: Apple LISA Image Centering

ElectricApple, 2021-07-11:
Absolutely brilliant project, great work.
I'm having difficulty getting the image centered from an Apple LISA using the latest software release and 12-bit board - see grey bar on right in photo.
It's a strange resolution of 720 with 176 blanking pixels x 364 with 15 blanking lines @ 20MHz 
I've tried tweaking all the geometry and sampling settings. 
When I increase the H offset too far the stable image becomes unstable and repeats across the screen.
Is there something else I can try?

[image]

IanSB, 2021-07-11:
Can you post a screenshot of the source summary page in the info menu and also use the save log function in the info menu and post log.txt from the SD card

ElectricApple, 2021-07-11:
[image]

[Log.txt](https://github.com/hoglet67/RGBtoHDMI/files/6797468/Log.txt)

IanSB, 2021-07-11:
It looks like the geometry settings are triggering a bug in the integer scaler, I'll have a look at that.
In the meantime a couple of things to try:
1. set the FB Size to Normal and see if that helps with adjusting the h offset (Dont forget to save the config before trying No.2 as that will cause a reboot losing your changes)
2. Change the scaling from integer to interpolated (keep FB size at Normal)

ElectricApple, 2021-07-11:
Thanks Ian. 

I've adjusted those settings and whilst looking better, I'm still unable to centre the image in the screen

IanSB, 2021-07-11:
Can you post your lisa.txt profile file

ElectricApple, 2021-07-11:
[LISA.txt](https://github.com/hoglet67/RGBtoHDMI/files/6797696/LISA.txt)

IanSB, 2021-07-11:
After reviewing the logs, it looks like the problem is the Hsync pulses which are 15uS long instead of the typical 5 uS (in the Lisa, video lines are about 44uS long).

I think the leading edge of the sync pulse is probably in the right place but the trailing edge which is used by the software to determine the start of the line actually occurs after the start of video so the left hand edge is getting cut off.

The latest beta already supports switching from trailing to leading edge sync pulse triggering so can you download beta36 from my fork:
https://github.com/IanSB/RGBtoHDMI/releases
Wipe the SD card. put the new files on and add your LISA.txt profile.

After rebooting, go to the sampling menu and change the sync edge from trailing to leading.
Then try adjusting the hoffset again.
Also 

Note this will require a CPLD update to v94 and if you revert to the old software you should manually downgrade back to the old CPLD version.

ElectricApple, 2021-07-12:
I've updated to the Beta release, and it's decreased the size of the grey bar and moved it over to the LHS, but unfortunately I still can't quite line up the image before it becomes unstable. I've attached screenshots of the output and the log file.

[image]
[image]
[image]
[LISA.txt](https://github.com/hoglet67/RGBtoHDMI/files/6798880/LISA.txt)
[Log.txt](https://github.com/hoglet67/RGBtoHDMI/files/6798881/Log.txt)

ElectricApple, 2021-07-12:
This may or may not be useful, but here's a capture of the video signals from PulseView 

[LISA Timing.sr.zip](https://github.com/hoglet67/RGBtoHDMI/files/6798950/LISA.Timing.sr.zip)

IanSB, 2021-07-12:
>This may or may not be useful, but here's a capture of the video signals from PulseView

Yes, that confirms my suspicions: The hsync pulse is actually wider than the blanking period.

[image]

In this case the leading edge starts before the end of the video line and the trailing edge ends after the start of the video line so there is no suitable edge to trigger capture of the video line itself, you will either get one end or the other end of the video cut off depending on which edge you trigger from.

Normally the hsync pulse sits entirely within the blanking period so you can trigger off either edge.

I notice that the Lisa monitor schematic refers to this signal as horizontal drive so it is not a standard hsync pulse:
https://www.applefritter.com/content/schematic-video-board-lisa-050-4012-h

Looking at the lisa schematic:
https://www.applefritter.com/content/schematic-cpu-lisa-5-5-050-4009-h
This shows !VSYNC, !HSYNC and !CSYNC going to J1 so it might be worth checking out the !CSYNC signal to see if it has different sized horizontal sync pulses that fit within the blanking period in which case that signal could be used instead. (You might have to experiment with changing the sync setting in the geometry menu although CSYNC could still be used as a form of HSYNC combined with the existing VSYNC. (Also worth checking the !HSYNC to see if it is the same signal you are picking up because it is inverted and your signal is non-inverted)

The other alternative would be to delay the hsync pulse by a few uS so that the leading edge fell within the blanking period. It might be possible to fit such a delay within the CPLD but it is rather full already.

ElectricApple, 2021-07-12:
Thanks again for looking into this Ian

I am able to centre up the image when using !CSYNC as the reference, my issue being that only H Drive, VSync and Video are presented to the video board connector which is located in the 'front' of the Lisa, with the remaining boards in the 'back'. 

I am currently taking these signals off the video board due to it's more convenient location for solder-free testing, meaning i've had to thread a wire through the back of the chassis to get a sniff of !CSYNC off a chip, which isn't ideal.  

Would it be possible to have an attempt at delaying HSYNC, as this would lead to a neater solution by only neededing to clip onto various points of the video board, unfortunately there's nowhere convenient to obtain !CSYNC from.

[image]
[image]
[image]
[image]
[image]
[image]

[LISA_CPU_CARD.sr.zip](https://github.com/hoglet67/RGBtoHDMI/files/6802896/LISA_CPU_CARD.sr.zip)

IanSB, 2021-07-13:
@ElectricApple 
>Would it be possible to have an attempt at delaying HSYNC

After some effort, I've managed to fit a limited delay option into the CPLD which should work with the Lisa:
[LisaSupportAlpha1.zip](https://github.com/hoglet67/RGBtoHDMI/files/6811794/LisaSupportAlpha1.zip)

As before, wipe your SD card and copy the files, adding your lisa profile
During first bootup, the CPLD will be erased so you can then program the v94_Lisa1 CPLD

After wiring up the separate H & V signals and selecting your Lisa profile, change the following settings in the sampling menu:
Clock Multiplier: x4
Sync Edge: Leading + Delay
Sample Mode 3 Bits Per Pixel 

(The delay only works in 3bpp mode as there was a spare combination of bits in that mode to select it - The video will still be 1pp as that is controlled by the 2 level mono palette) 

Then change these Geometry Settings:
H Pixel Aspect: 2  
V Pixel Aspect: 3 
FB Size: Double Height
Sync Type +H+V

And finally adjust the H Offset to centre the image.

The added delay is 255 sample clock cycles where the sample clock is the pixel clock times the multiplier setting 
So with a pixel clock of 20.373351Mhz x4 that gives approx 3.12uS of delay

After setting up, you can try the x6 multiplier which gives 2.086uS of delay which is very marginal but might still work.
(Based on your PulseView timings, the minimum delay required is around 2.05uS)
Any multiplier above x6 will not give enough delay.  

The above aspect settings will ensure correct integer scaling on a 1600x1200 or 1920x1200 monitor but a 1080p will still look a bit letterboxed as there aren't enough vertical pixels for 3x integer scaling so interpolated will look better.
You could get integer scaling working on a 1080p monitor by reducing the vertical capture by 4 lines (360x3 = 1080)

If you get a good result, please post your profile and I'll add it to the next release

ElectricApple, 2021-07-13:
Ian
Thanks for your sterling efforts, they are much appreciated especially as the tube in my Lisa is starting to wear out considerably.

I can confirm that your additional delay has allowed the Pi to sync correctly to the Video board signals and allow the image to be centered! It works at both x4 and x6 multipliers as suggested. 

I've included two profiles, one for integer and one for interpolated.
I personally prefer the interpolated so will probably stick with that.

Thanks again!

[Apple_Lisa.zip](https://github.com/hoglet67/RGBtoHDMI/files/6812101/Apple_Lisa.zip)

IanSB, 2021-07-13:
@ElectricApple 

Thanks for the profiles.

Can you do a few screencaps for the gallery using the built in screencap function (short press of SW2, files on SD card)

ElectricApple, 2021-07-14:
No problem. Here's a few to choose from
[Apple_Lisa Captures.zip](https://github.com/hoglet67/RGBtoHDMI/files/6813626/Apple_Lisa.Captures.zip)

closed 2021-07-14 by ElectricApple

IanSB, 2021-07-25:
@ElectricApple 

All those changes are now available in beta 37 release:
https://github.com/IanSB/RGBtoHDMI/releases

This contains a single "Apple Lisa  TTL" profile that should automatically select the height as appropriate so that integer scaling works reasonably well on a 1080p monitor.

i.e.
with integer scaling on 1080p you will get 360 lines captured (2 missing from top & 2 from bottom)
with interpolated scaling on 1080p you get the full 364 lines
with integer scaling on 1600x1200 or 1900x1200 monitors you will also get the full 364 lines

Can you check that works OK for you.

BTW I saw this video on Youtube about the Lisa2 / Mac XL which shows a composite video output on  the back:
https://youtu.be/7TOh_W6EUOc?t=200
It should be possible to use that output directly with the analog addon board with identical quality to the internal TTL pickup so I am going to add a speculative profile to support that as well although it will likely need some tweaking.

Is there such a connector on your Lisa?

ElectricApple, 2021-07-26:
Ian

Thanks for tweaking the modes to auto adjust the height etc. 
Appears to work perfectly on the Lisa 

My machine does indeed have a video out socket.
No idea how I forgot about that, it would make a lot more sense!
I haven't made up any analog boards, but I have a couple of other machines I'd like to test out so when i get around to making one i'll try the analog Lisa profile and report back.