hoglet67/RGBtoHDMI #223: Question on IIe NTSC support

timmcross, 2021-07-17:
I wired up a 6/8 board to a NTSC IIe as described in the cabling section. However, I didn't see a profile for Apple IIe.  So should RGB IIc be used?  With IIc under beta36, the colors weren't right.  I played with the settings and I think I corrected it.  Let me know if there is another profile I should use or if you would like this profile.  Should I use a 12 bit board?

IanSB, 2021-07-17:
Yes, I haven't added a TTL IIe profile yet so use the IIc profile. If you had to change the Artifact Colour Phase to get the right colours using the IIc profile, please post your saved profile and I'll add that (assuming it is a different value from the IIc) 

Note since beta 34 there is a new feature that pressing SW3 when Artifact colours are enabled will cycle through the 4 phases instead of having to use the menu. This option is mainly aimed at the Tandy CoCo machines where the phase varies by 180 degrees randomly on reset so it is a quick way to fix wrong colours but it works with any computer with artifact colours.

timmcross, 2021-07-26:
Here is what I found for the IIe.  I have two to test with (different motherboard revs).  I started with the IIc profile and modified it until the colors looked correct.  Brown still seems a little green but everything else seems right.  Hopefully this will help.  Let me know if you would like a screenshot of any particular screen.
[Apple_IIE_TTL.txt](https://github.com/hoglet67/RGBtoHDMI/files/6875364/Apple_IIE_TTL.txt)

IanSB, 2021-08-04:
That's been added to the profile set