hoglet67/RGBtoHDMI #225: Feature request : option to add a mute audio channel

pixelk0, 2021-07-21:
Hello,
I wanted to use the RGB2HDMI with a Monoprice Blackbird Audio Inserter (p/n 13347) but this device can only replace an existing audio channel, not add one when there is none (according to a few amazon review, that I read after buying it, of course). Would it be possible to add an option to ouput a silent audio track along the video on the HDMI output, to compensate fr this shortcoming (that is absolutly not on RGB2HDMI's side, I'm well aware) ?

IanSB, 2021-07-24:
Have you tried setting the HDMI mode to "HDMI ..." rather than DVI compatible?

After setting that, try editing the config.txt file on the SD card to add the following under the "# Common video settings" section:
hdmi_force_edid_audio=1

pixelk0, 2021-07-25:
It works ! Thank you !

closed 2021-07-25 by pixelk0

IanSB, 2021-07-25:
@pixelk0 

Did just setting to the HDMI output work or do you need the hdmi_force_edid_audio=1 as well?
(If the latter I'll add that to a future release as part of the HDMI output config)

pixelk0, 2021-07-26:
It works with or without hdmi_force_edid_audio=1 in config.txt, but not without the switch from DVI to HDMI(Auto RGB/YUV). I tested all 4 combinations.

psypoker, 2021-09-05:
Thanks mates ! searching for hours :) amiga at the best