hoglet67/RGBtoHDMI #226: Flash Floppy OSD on 12bit board

Croccy22, 2021-07-26:
I can't find a definate answer to this. Does the flash floppy OSD input on pin 18 work on the 12bit board? Or does it only work on the copperdragon board for the Amiga?

I've looked at the 12bit board and it appears pin 18 actually links to the CPLD so is probably used for something else?

Can someone confirm?

Thanks, Matt.

IanSB, 2021-07-30:
Pin 18 does work with FFOSD on the 12 bit board but only in 12 bits per pixel mode. This is because the pin is sampled in software during video capture and it can only sample at the pixel rate in 12bpp mode. In 1,3 or 6 bpp mode the CPLD buffers the pixels and transfers them at a lower data rate so the software would only be able to look at the FFOSD input every 8, 4 or 2 pixels respectively.
Pin 18 is also used to program the CPLD which is why the FFOSD signal must be connected via a 1K resistor so it can be overridden during programming.

Croccy22, 2021-07-30:
Thanks for the reply. I assume this means it won't work with the Sam Coupe profile with that being 7bit?

G4DDS, 2021-08-15:
I have one of Croccy22's board for my SAM Coupe and have been in (private) contact with Croccy22 to discuss the use of the OSD pin.

Currently, it is linked up to pin 18 using a 1K Ohm resistor and I managed to to get the OSD overlay working using an Amiga profile (just to test that the OSD was working). The next step was to switch to the supplied SAM Coupe profile and make some adjustments to enable the OSD:

Sampling Menu - Sample Mode changed from 9 Bits to 12 Bits Per Pixel.
Geometry Menu - FB Bits/Pixel changed from 8 to 16.

This resulted in a horizontally squashed screen so I played around with the width setting and changing this appeared to fix the aspect ratio:

Geometry Menu - Max H Width changed from 608 to 728

However, not knowing enough about how it all works I don't know whether this is correct or it was just a fluke and there is something else I could (or should) have changed. Any thoughts?

The real problem I have though is that the palette seems to be off with a pinkish tinge since swapping to 16 FB Bits/Pixel. Is there an alternative palette profile that can be used to fix this?

On my setup I created a SAM_Coupe directory containing two profiles called Sam_Coupe_FF_OSD and Sam_Coupe_No_FF_OSD to allow me to easily swap between the two profiles for maximum flexibility especially as there is no easy way to enable and disable the OSD display. It seemed odd to me that there is no option in the settings menu for this?

IanSB, 2021-08-15:
@G4DDS 

The main problem at the moment is that it is very difficult to provide a general purpose capture loop that works with all options at all bit depths as that causes performance issues so in many cases separate capture loops are provided for separate options and combinations of options but this means that the number of capture loops required can increase exponentially as the number of options increases so only capture loops required for supported profiles are implemented and so some options like FFOSD only work at certain bit depths.

One longer term solution would be to create additional capture loops to support those options at other bit depths but working around it like you are trying by using the 12bit into 16 bit frame buffer is another viable solution.

The reason for the pink tinge is that the intensity bit is only connected to the R1 input but that adjusts all RGB levels on the SAM Couple palette. When you use the 16 bit frame buffer there is no palette and the RGB values are used directly so the intensity bit is only affecting the R1 input, hence the extra red added to all high intensity colours. To get more correct colours you have to connect the Intensity bit also to the B1 and G1 inputs so it affects all 3 simultaneously. You can achieve this by connecting R1 to B1 (adjacent pin 2 on P2) and G1 (Pin2 on the P5 connector holes at the other end of the board next to R0,G0,B0)

Note the direct colours in 12bpp mode will be dimmer than the SAM Palette:
In the palette:    bit 3 = 0x92, bit 2 = 0x49, bit 1 = 0x24 (So all bits on = intensity 0xff or 255)
In 16bpp direct: bit 3 = 0x88, bit 2 = 0x44, bit 1 = 0x22 (So all bits on = intensity 0xee or 238)

If you connected the intensity bit to R0,G0,B0 as well then the maximum level would be boosted back to 255:
In 16bpp direct: bit 3 = 0x88, bit 2 = 0x44, bit 1 = 0x33 (So all bits on = intensity 0xff or 255)
Closer but not an exact match for the SAM palette.

>no easy way to enable and disable the OSD display. It seemed odd to me that there is no option in the settings menu for this?

Beta 40 in my fork has FFOSD on/off in the menus
https://github.com/IanSB/RGBtoHDMI/releases

It also has an option for two sets of sampling/geometry settings which can be toggled by pressing SW3 so you could toggle between the standard profile and the the above tweaked on if the different palette was objectionable.

Not sure about the aspect ratio change, I'll have to look into that.

G4DDS, 2021-08-15:
That's given me something to think about. To try and get the SAM palette correct (and see whether it is really that much dimmer that it is noticeable) I do have Xilinx ISE Design Suite 14.7 installed (for messing around with FPGA boards) which I believe can be used with the VHDL projects to rebuild the CPLD files. Looking at the VHDL sections that reference B1_I and G1_I  we have:

shift_G <= R2_I & G3_I & G0_I & B1_I;             -- 12 bpp

shift_B <= R3_I & R0_I & G1_I & B2_I;             -- 12 bpp

Would replacing the occurrences of B1_I and G1_I with R1_I be the equivalent of doing the hardware mod you described to link R1, G1 and B1 together?

I also notice that B1_I is also used to generate signals B2 and B3:

    B3 <= B2_I when B1_I = '1' else B3_I;
    B2 <= B3_I when B1_I = '1' else B2_I;

But here I think I can leave these alone as B2 and B3 appear to be used to in the 6 bpp profiles only.

I have just seen your edit about the aspect ratio problem - it's too late to digest but looks like I have worked around it anyway.

IanSB, 2021-08-16:
@G4DDS 
>Would replacing the occurrences of B1_I and G1_I with R1_I be the equivalent of doing the hardware mod you described to link R1, G1 and B1 together?

Yes, in fact you should replace R0_I , B0_I and G0_I with R1_I as well in all three shift_R/G/B lines to get the best result with intensity more closely matching the  original as suggested above.

>I also notice that B1_I is also used to generate signals B2 and B3:

That is only used in analog mode as part of a 3 line to 2 bit encoder and not used in normal digital capture

Here is a slightly tweaked version of the code which fixes the width issue and also does some internal bit manipulation on R1 and B1 as I think the standard SAM profile might not work properly once B1 gets tied to R1 with the old software.
Download Beta40 from here
https://github.com/IanSB/RGBtoHDMI/releases
Then overwrite the kernel with the file in this zip:
[kernelrpi.zip](https://github.com/hoglet67/RGBtoHDMI/files/6989737/kernelrpi.zip)

If you are using the above software then download the latest RGB VHDL (V9.4) from the dev branch

After that you just need to change sample mode to 12bpp and FB bit depth to 16bpp and tie R1 to B1,G1,R0,G0,B0 either physically or by changing the CPLD.
If that all works I'll include those updates in the next stable release due soon. 

I looked at adding FFOSD to 8bpp frame buffers and the bit detection is simple enough but it also requires the palette to be dynamically reprogrammed when FFOSD is detected and restored when the signal is dropped and that is a little more complex due to some timing issues adding that code in the relevant place, so if the above works it is the best option for now.

G4DDS, 2021-08-16:
I don't think the FF OSD setting is working as intended - with it off the OSD is not displayed but the screen still dims when it detects the OSD on pin 18.

I made the changes to the 12 bpp sections of the VHDL (took me a while to work out how to generate a xsvf file):

shift_R <= R1_I & G2_I & B3_I & R1_I;             -- 12 bpp
shift_G <= R2_I & G3_I & R1_I & R1_I;             -- 12 bpp
shift_B <= R3_I & R1_I & R1_I & B2_I;             -- 12 bpp

And this took the pink tinge off and the palette looked OK to me.

One thing though I had an error during the fitting stage saying it had run out of space so I removed the 1, 3 and 6 bpp settings (as it was just an experiment anyway) just to get it to generate. I will leave it upto you to work out how to fix that but thinking about it would you not need a specific 12 bpp setup for the SAM Coupe profile anyway (i.e. a specific CPLD firmware just for the SAM)?

IanSB, 2021-08-17:
@G4DDS 
>I don't think the FF OSD setting is working as intended

Yes that was a bug which I have now fixed

>One thing though I had an error during the fitting stage saying it had run out of space

The CPLD is very full with not much room for even minor tweaks

Anyway I have now managed to figure out a simple way to update the palette for FFOSD to work with 8bpp frame buffers so here is an updated kernel which should work with the original SAM profile and standard CPLD, no hardware changes required:
[kernelrpi.zip](https://github.com/hoglet67/RGBtoHDMI/files/7001074/kernelrpi.zip)

Overwrite the existing kernel with this version, reprogram the CPLD back to the default 9.4, restore the profile back to default, switch FFOSD on in the settings menu and see if it works.

G4DDS, 2021-08-18:
@IanSB 

I started again with Beta40 and the latest kernel from the previous post, programmed v9.4 in and I am happy to say it all looks as though it is working with the standard SAM profile (9 bpp sampling and 8 FB Bits/Pixel). I can confirm that the FF OSD setting has also been fixed.

On a un-related point but still to do with the SAM profile (so if an update is needed it could potentially be included in the next release), when using the SCART output the border on the left and right is a lot wider than when it is output on the HDMI. I have attached a couple of screenshots. Is there a simple tweak needed to the profile that can give us a wider border?

[image]
[image]

IanSB, 2021-08-18:
@G4DDS 
>Is there a simple tweak needed to the profile that can give us a wider border

Try this version:
[kernelrpi.zip](https://github.com/hoglet67/RGBtoHDMI/files/7010028/kernelrpi.zip)

With a 1920x1080 monitor, more than the 4:3 section (which would be 1440x1080) can be used to zoom in slightly if the top and bottom borders are unused for normal video which is the case with most computers that have 192 or 200 pixels vertically.
The extended width was limited to 1600x1080 but I have now increased it to 1800x1080 and that should result in an increased border but it is untested. (This is all due to the SAM having a non-square pixel aspect ratio which scales by odd numbers compared to most computers so fitting it with integer scaling is difficult)

BTW  This uses integer scaling for a very sharp image but check that your TV is configured for 1:1 pixel mapping otherwise the sharpness will be lost as many TVs will zoom into the image by 5% or so by default. The name of the option varies depending on TV manufacturer and should be in with the aspect ratio controls but may be something like "just scan", "exact scan" "PC mode" etc although some TVs don't have the option at all and you are stuck with a permanent zoom.

G4DDS, 2021-08-18:
[image]
[image]

That looks better what do you think @Croccy22 - maybe you could check this out on your setup and see what you think? By the way I had to change the Horizontal offset to 120 so the image was more central so a little update to the profile in the distribution may be needed.

IanSB, 2021-08-19:
@G4DDS 
>I had to change the Horizontal offset to 120 so the image was more central

I increased the width slightly more to 1824x1080 which fixes a rounding error that resulted in uneven black borders in the above.
Please try this version:
[kernelrpi.zip](https://github.com/hoglet67/RGBtoHDMI/files/7011535/kernelrpi.zip)
It should be centred correctly although you may need to further adjust the h offset (in geometry) and pixel h offset (in sampling) to get the image exactly centred to pixel accuracy.

G4DDS, 2021-08-19:
[image]
[image]

I left the horizontal offset at 120 and offset at 6 - I think it looks OK. Also attached the profile file I am using in case you want to put that in the next release (it has the FF OSD setting in I see as well).

[Sam_Coupe_update.txt](https://github.com/hoglet67/RGBtoHDMI/files/7016587/Sam_Coupe_update.txt)

G4DDS, 2021-09-01:
@IanSB 

Just wondering when you think the next stable release is likely to be?

IanSB, 2021-09-01:
@G4DDS 

Beta41 in my fork is a release candidate for the next stable version so if all goes well it should be soon
https://github.com/IanSB/RGBtoHDMI/releases

Please give it a check with your system and report back.

G4DDS, 2021-09-02:
@IanSB 

I tried beta 41 and it all looks OK to me.

One thing though and I may add I saw this on the original stable release software is that very occasionally I get some distortion/corruption of the bottom part of the screen that covers the final bottom few pixels of the actual screen and I assume the border as well (but since it's all one colour won't be noticed). It only does it very occasionally and only appears on screen (the captured images are fine) and switching the SAM off and on is normally enough to fix it. Do you have any ideas what might be the problem?

[image]
[image]
[image]
[image]

Croccy22, 2021-09-02:
Have you tried on a different TV/Monitor? I've never seen this issue on my TV and have obviously tested a lot of interfaces :)

G4DDS, 2021-09-02:
I have only seen it on my smaller HD ready, not on my 43" LG but I have noticed that there is also some interference pattern at the bottom right:

[image]
[image]
[image]

I don't know why I didn't try this before but I have just discovered that power cycling the RGBtoHDMI interface (and leaving the SAM powered up on the above screen) also appears to solve the problem.

IanSB, 2021-09-02:
@G4DDS 

Those lines look like they are only one HDMI pixel high and the software doesn't generate pixels that small. The frame buffer stores the source pixels at much lower resolution (either the resolution of the source computer or double the resolution depending on the FBsize option) and they are scaled up by the GPU.

A likely explanation is that the signals on the HDMI interface are marginal resuting in some sync issues especially if you are using a mini HDMI adapter rather than a lead with the correct connectors and resetting the Pi forces the HDMI interface to reinitialise so try another HDMI lead and/or adapter or even another Pi zero.

You can also try increasing the HDMI signal strength by adding a line to the config.txt file under "Common video settings":

config_hdmi_boost=7

Valid values are 0-11 and it defaults to 5. Do not use very high values unless absolutely necessary.