hoglet67/RGBtoHDMI #232: No Issue. CPLD Preference question

akldjian, 2021-08-17:
Hi everyone,
I'd like to ask what the preferable CPLD is. The BOM lists the XC9572XL-10VQG44C as the primary CPLD. The alternative is the XC9572XL-7VQG44C. The alternative seems to be faster and has less delay. From a long term performance perspective, is the alternative the preferred choice? The only downside I see for the alternative is the cost, but it's not significantly more. Thank you.

IanSB, 2021-08-20:
The faster part might guarantee that it will work at a higher frequency but I have found that both parts seem to fail at around the same overclocked frequency with this application.

akldjian, 2021-08-20:
> The faster part might guarantee that it will work at a higher frequency but I have found that both parts seem to fail at around the same overclocked frequency with this application.

Do we know if the XC9572XL-5VQG44C is supported? This is yet even faster than the previous two mentioned. Same package and power. For reference, the 10VQG44C=100MHz (10ns latency), 7VQG44C=125MHz (7.5ns latency), and the 5VQG44C=178.6MHz (5ns latency).

Does the software have/require provisions for each of these CPLD variants? Thank you.

IanSB, 2021-08-21:
>Do we know if the XC9572XL-5VQG44C is supported?

I've not tried one but I can't think of any reason why it shouldn't work.

>For reference, the 10VQG44C=100MHz (10ns latency), 7VQG44C=125MHz (7.5ns latency), and the 5VQG44C=178.6MHz 

Every 10ns one I have tried (about 80 so far from multiple batches over a year or so) has worked up to 178 Mhz so all the speed grades are probably the same die, just graded according to reliability tests. Maybe the firmware for this project doesn't stress the CPLD that much or perhaps the production yield is so good that virtually every part could work as a 5ns part at least at normal room temperatures, perhaps the slower rated parts don't perform as well over the full specified temperature range.

I did try 20 of the 7ns parts but there was no real difference in the overclockability compared to the 10ns parts and as I got much better than expected results with the slowest parts I haven't bothered to investigate the faster parts any further. Some of the 7ns or 10ns parts will even work up to 200Mhz but the 7ns parts were not any more overclockable than the 10ns ones.

>Does the software have/require provisions for each of these CPLD variants

No software difference needed.

closed 2022-12-20 by IanSB