hoglet67/RGBtoHDMI #233: single_button_mode not working in release 20210529_777ea4f

maximevince, 2021-08-19:
Hi,

First of all: congrats with the incredible project. This is just mind-blowing.

Now for the issue: I am ussing c0pperdragon's Amiga-Digital-Video board,
but with the latest firmware release `20210529_777ea4f`, I am unable to get the `single_button_mode=1` working.
When I revert back to release `20210322_f771e51`, it works as expected.

Is this a known issue, or am I doing something wrong?

Cheers!

IanSB, 2021-08-20:
@maximevince 
>but with the latest firmware release 20210529_777ea4f, I am unable to get the single_button_mode=1 working.

Setting single_button_mode=1 puts a c0pperdragon board into 3 button mode.

I just tried this on a c0pperdragon board and it seemed to work OK for me with build 20210529_777ea4f and grounding the appropriate three inputs worked OK.

What exactly are you seeing?

BTW make sure that there is no space between the = and 1 when you edit default.txt as that can cause problems.

bkodenkt, 2022-07-15:
I experienced the same issues. No matter where I set single_button_mode=1, the single button only worked for switching scanlines on/off and opening/closing the menu. After a lot of trying, reverting to the release mentioned above instantly fixed the problem.

maximevince, 2024-04-10:
For anyone still wondering. For some reason `single_button_mode` is inverted for `cpld_version == DESIGN_SIMPLE`, i.e. all the early hardware designs without a CPLD.
See: https://github.com/hoglet67/RGBtoHDMI/blob/master/src/osd.c#L5434

I have no idea what the reason is for this. But setting `single_button_mode=0` effectively _enables_ single button mode on the non-CPLD versions...

It has been introduced in this commit: https://github.com/hoglet67/RGBtoHDMI/commit/7d438d385c86392fa441b5503b9f9ab4e014c66f