hoglet67/RGBtoHDMI #234: Analog - Amiga support?

chrisblown, 2021-08-30:
This is not an issue specifically - sorry if putting this here is taboo, not sure where else to put it. I am seeing a lot more support for analog RGB signals and while I understand the Amiga Denise chip solution grabs the digital signals, could this project convert the Amiga's existing analog RGB (from the 23pin). 

My specific application is a custom Video card module for a Commodore CDTV that replaces the existing Comp / RF card module - such a card could mount the Pi and be 100% plug and play without needing to open the CDTV. The IO pins on the Video slot contain all the analog sync and RGB outputs. (As per the service manual).

hoglet67, 2021-08-30:
The current RGBtoHDMI analog board is only able to descriminate 3 or 4 discrete voltage levels on each of R, G and B, because it uses simple comparators rather than a high precision video ADC.

The analog output from the CTDV (and the Amiga) has 12-bit colour (4096 colours),  which gives 16 different discrete voltage levels on each of R, G and B.

Sorry, but there is no way this is going to work well.

That's why all the effort on by the Amiga folks has focussed on tapping in to the 12-bit digital interface.

Dave

chrisblown, 2021-08-30:
Thx Dave, ok got it.