hoglet67/RGBtoHDMI #236: Colour issue with Amstrad CPC464

Imperious685, 2021-09-03:
I've just built one of these RGBtoHDMI with a version 4 12 bit board and version 3 Analog board. So far I have only tested with an Amstrad CPC464 but have some Colour issues. First of all I have checked thoroughly that all components are correct values and soldering is good with no bridges or dry joints. The CPC is good as it is all ok with my OSSC. I've tried every adjustment and nothing seems to fix this. Maybe a faulty chip or even one of the pcbs. I have yet to do a full circuit test with multimeter.  I have attached photos. Someone at CPCwiki said it looks like CGA except RGB are all there just wrong in places. Any help much appreciated.
 
[image]
[image]
[image]
[image]

hoglet67, 2021-09-03:
Can you confirm the above photos are with a fresh installation of the latest stable software (20210529_777ea4f)

Do you think you could post screen captures saved by RGBtoHDMI instead of photos?

Some more infomation would be helpful:
- A log file (the option to save one is in the INFO menu)
- A photo or screen capture of your Info/Source Summary Menu
- A photo or screen capture of your Info/System Summary Menu
- A photo or screen capture of your Palette Menu
- A photo or screen capture of your Samping Menu

hoglet67, 2021-09-03:
The Amstrard has three levels for each of R, G and B: 0%, 50% and 100%, giving 27 total colours.

On your photos it looks like the 50% and 100% values for each of the colors are being merged together, the end result being the images only contain 8 colours.

One thing that would cause this is if there was a problem with the programmable 75R termination circuit on the analog board. The Amstrad CPC profile enables termination, and this is needed to get the required voltage levels to match the profiles. If the termination was always off, then that might cause a problem like this.

Schematic is here:
https://raw.githack.com/hoglet67/RGBtoHDMI/master/kicad_analog_6bit/V3/analog.pdf

I'd be checking U3, C9 and R5 as they are common to all channels.

Imperious685, 2021-09-03:
Thanks Hoglet67. I'll do all that checking on the Analog board tomorrow and report back. If I can't find a problem there I'll post some screengrabs as well. The 75R termination does work as I've already tried it off and on and it looks terrible off.  
I have used the latest software (20210529_777ea4f)

hoglet67, 2021-09-03:
Thanks. Please make sure when take the screen shots that all the settings have their default values. Ideally start by wiping the SD card and re-installing the software.

Imperious685, 2021-09-05:
I've had a good look at my Analog board and did find a short between pins 1 and 2 on U4, once removed made no difference at all. One more likely thing I should mention is that due to lack of availability I used a CD4066BM instead of a 74LVC4066. I'm going to say this could be the problem as the LVC part is designed for lower operating and input voltages. The CD4066BM is at the lower limit of operating at 3.3v. The only other thing Is I have not installed U7 but nothing I can find tells me that I have to add that with a 12bit board?

hoglet67, 2021-09-05:
> One more likely thing I should mention is that due to lack of availability I used a CD4066BM instead of a 74LVC4066.
Yes, this would have been good to meantion earlier ;-)

The CD4066BM on resistance is typically 470 ohms (at 5V), higher still at 3.3V.

The 74LVC4066 on resistance is is typically 6.5 ohms (at 3.3V)

The 75R termination is not going to work well with a CD4066BM, which is what's screwing up the levels.

That's almost certainly the issue here.

Dave

hoglet67, 2021-09-05:
What country are you in?

hoglet67, 2021-09-05:
DigiKey have these in stock:
https://www.digikey.com/en/products/detail/nexperia-usa-inc/74LVC4066D-118/2763168

Imperious685, 2021-09-05:
Thanks Hoglet. I'm in Australia so might just order from Aliexpress and hope for the best. My Bad with that part, I was paying more attention to the pinout when ordering. The LVC parts not available until 2022 from RS components here. I can get 74HC4066 but expect that won't cut it either. There are some other Aussie based suppliers so will check them out. 
Well I guess if anyone else makes this mistake the answers here.

hoglet67, 2021-09-05:
DigiKey Australia show them in stock as well.

hoglet67, 2021-09-05:
Are you only using this with the CPC?

If so, you could just short out the termination switches:
- Short U3 pins 3 and 4
- Short U3 pins 8 and 9
- Short U3 pins 10 and 11

That will permenantly enable termination.

Imperious685, 2021-09-05:
Thanks I'll do that.  Digikey Aus and Mouser Aus all parts come from USA.

IanSB, 2021-09-09:
@Imperious685 

The resistance of the analog switches is added to the 75R resistor when termination is enabled so you were terminating in several hundred ohms instead of about 81R with the correct chip. The analog RGB output output of the CPC relies on the termination resistor as part of it's output circuit and if that is very wrong or missing then all the RGB voltages will be wrong.

>Atari 2600 with RGB mod

There isn't a profile for the Atari 2600 with RGB mod so you will have to create one.
Also I think the 2600 has 128 colours and the analog board can only distinguish 27 colours or 64 colours with the U7 fitted.
It may be possible to get a usable output by setting up your profile but some colours will be a shade wrong.

> ZX spectrum +2A and possibly my RGB modded C64

Both of these also requires termination so should be OK with hoglet's workaround above (assuming the c64 is using c0pperdragon's mod).

Imperious685, 2021-09-10:
Thanks IanSB.
I've ordered the correct 74LVC4066 ic's from Aliexpress, until them shorted pins will suffice.
It's quite ironic one of the oldest consoles has 128 Colours. Think I'll just try see how I go with the OSSC, otherwise I have a GBS8200 that I can use with the Control mod. It ends up being a case of Horses for Courses, some adaptors are better for certain applications. My C64 does have the Copperdragon mod.

closed 2021-09-10 by Imperious685