hoglet67/RGBtoHDMI #239: Composite out on Raspberry pi's with composite out 3.5mm jacks.

qbancoffee, 2021-09-21:
I know this seems like an odd feature request but it can't hurt to ask.....

So I have a bunch of raspberry pi's lying around of varying models and they all have composite out.
This is something I have used in the past to add a touch of authenticity to the retro experience and it works well enough
with some composite CRT monitors I have used and even CRT television sets. 

The raspberry pi shares the composite video out with the 3.5mm jack and it is usually a matter of making one or two changes to the config.txt file.

I tried enabling composite out on the pi and There is a rainbow splash screen that is displayed on the composite out but then it keeps rebooting.

Is this something that is disabled by this project? Is there a performance impact by using composite out?

I know this request runs counter to the intent of this project but I figured since the option to use composite out is baked into
the raspberry pi why not use it as well?

In my case I have serveral TRS-80 color computers that do not have composite out and and if the RGB to HDMI solution can output both HDMI and composite out, it is more valuable to me and hopefully others. I know I would still like to use my composite CRT monitor for some things.

qbancoffee, 2021-10-08:
Is there an setting in the config.txt to disable rebooting? 
Is this something that has to be changed in the code and recompiled?
Can this functionality be added by means of setting a variable in the config.txt?

IanSB, 2021-10-09:
@qbancoffee 
>Is there an setting in the config.txt to disable rebooting?

not yet...

>Is this something that has to be changed in the code and recompiled?
>Can this functionality be added by means of setting a variable in the config.txt?

Try this kernel version which has rebooting disabled:
[noreboot-alpha1.zip](https://github.com/hoglet67/RGBtoHDMI/files/7316299/noreboot-alpha1.zip)

This is untested so there may be other issues that have to be fixed when using such a low resolution.
Start with the latest beta or release version (they are the same) and overwrite the kernel of that.

IanSB, 2021-10-10:
@qbancoffee 

It won't work with the Pi 3 as that requires a different kernel file so it's just using the unchanged pi3 kernel 

The narrow width is due to the internal readback of the horizontal size being doubled in very low res modes
Try this version which attempts to compensate for that:
[noreboot-alpha2.zip](https://github.com/hoglet67/RGBtoHDMI/files/7317824/noreboot-alpha2.zip)

qbancoffee, 2021-10-10:
With this kernel the width is normal now but without color.

IanSB, 2021-10-12:
@qbancoffee 
>With this kernel the width is normal now but without color.

I don't have a system setup to look at the composite output so it may be some time before I can investigate this but it might be worth you experimenting with config.txt settings as documented here:
https://www.raspberrypi.com/documentation/computers/config_txt.html#composite-video-mode
(Add any new options to the common video settings section of config.txt)