hoglet67/RGBtoHDMI #240: No Issue, just question/proposal (fringe use case)

vectorjoe-hh, 2021-09-30:
Hi, first of all thank you for your great work, and sorry if this is the wrong place for a suggestion, please delete in that case.

I am currently consolizing PSP´s with defective display, meaning i turn them into stadionary consoles, since the -2000 and-3000 series of the PSP´s support TV component out via a propritary connector. The existing AV cables in combination with component to hdmi converters result in a suboptimal image quality with black borders around all 4 sides. 
Would it in theory be possible to use RGB2HDMI on PSP´s? I know it only makes sense if you turn your psp into something stationary, and i know that this is a very fringe use case. I am willing to donate a working psp with defect display to a developer.

https://manuals.playstation.net/document/en/psp/current/settings/aboutvo_3000.html
[image]

IanSB, 2021-10-08:
The PSP is 24 bit colour and RGBtoHDMI only supports 12 bit colour so this isn't going to be practical.

vectorjoe-hh, 2021-10-09:
What a pitty. Thx for the answer.

closed 2021-10-09 by vectorjoe-hh