hoglet67/RGBtoHDMI #244: Analog Board on C64

lolof70, 2021-11-02:
Hello,

I soldered an analog board for my C64 (with FPGA video mod).
It does not work well, missing colours or something like that.
I built the board following the BOM made by Kicad. 
Trying to get the board working, I saw a bom on the wiki right now...
There are not both the same (both boms).... 
On my board : 
- R1 is 100 Ohm, 
- U6 is TS5A3159ADBVR
- U1 and U2 are MAX9108ESD+
All those are coming from Kikad BOM.

Can it be that I have a problem with the analog board because one or more of those are not correct ?

Here pictures from the RB2HDMI in action.

[image]
[image]
[image]
[image]

lolof70, 2021-11-03:
Just put R1 with 470 Ohm. It does not help.... :-(

lolof70, 2021-11-03:
Oh yea.....

Problem solved. When I built my fpga video board for the C64, I replaced 2 resistor with lower value one  to reduce the colours saturation....
I put the original value back and all is working. Perfect !

I also tested with the bridges on U3 removed. Still work perfectly.

Thank you for this very nice adaptor.

IanSB, 2021-11-04:
@lolof70 
Sorry, I only just noticed your post, glad you got it sorted out already.

As you discovered, the RGB values have to be at predictable levels which are set either by reprogramming the colour tables (with older c0pperdragon firmware) or by putting the fpga board into the special RGBtoHDMI mode (with newer c0pperdragon firmware).

The analog board level slices the RGB values to extract the bit-exact digital data from the C64 and by changing the resistors you messed up the digital extraction.

>I built the board following the BOM made by Kicad.
Trying to get the board working, I saw a bom on the wiki right now...
There are not both the same (both boms)....

The wiki is correct, the values got changed (to improve performance or because components were not readily available) without requiring a PCB update so the kicad files didn't get updated

lolof70, 2021-11-04:
Thank you very much for the reply.
I must say, it's working very good with the bom from Kicad and R1 with 470 Ohm.
It's just perfect, at least with the C64..
Thanks for all.