hoglet67/RGBtoHDMI #245: quad core pi zero

rogerjowett, 2021-11-03:
https://www.tomshardware.com/uk/news/radxa-zero-sbc-pi-zero-clone

this is a quad core pi zero
raspberry pi is also making quad core pi zero
please would it be possible to use this in an rgb2hdmi interface and simultaneously encode the video output to mpeg2 avi video file? 
rogerjowett@gmail.com
Please let me know what you think

hoglet67, 2021-11-03:
Sorry, this won't be possible.

hoglet67, 2021-11-03:
Encoding MPEG2 video is currently out of scope for this project.

closed 2021-11-03 by hoglet67

rogerjowett, 2021-11-03:
do you know absolutely anyone for whom it might be in scope please?

hoglet67, 2021-11-04:
Roger, please stop spamming this project with off-topic question, otherwise I will be forced to block you.

hoglet67, 2021-11-04:
Seems you have a bit of a reputation for spamming!
https://www.cpcwiki.eu/forum/general-discussion/who-is-roger-jowett/