hoglet67/RGBtoHDMI #249: Atari TT030 ECL support

Sch-LikA, 2021-11-08:
I am refurbishing an [Atari TT030](https://en.wikipedia.org/wiki/Atari_TT030) and this machine can output a high resolution monochrome 1280×960 'ECL' mode. This mode is required to run the unfortunate Atari Unix ...

I am building [this](https://github.com/tenox7/tenoxvga) adapter to convert ECL to VGA, but would it be possible to implement ECL support in rgb2hdmi ?

Thanks again for the great project !!

Sch-LikA, 2021-11-10:
Found a few details about ECL [here](https://web.archive.org/web/20000919004010/http://www.hut.fi/~pelle/vga-ecl.txt)

'Viking video signal is ECL level -- i.e. -5 to 0 V.  It is also
complementary (video signal has separate plus and minus signals.
The ECL input acts as an comparator.  The result is on or off 
depending on which signal has higher potential.

Both plus-video and minus-video have their on ground wires.

As ECL input is a comparator we can fool it to accept video signal.'

IanSB, 2021-11-20:
I think the pixel clock is going to be around 96 mhz for that sort of resolution and that is too high for the CPLD to sample.
The CPLD normally runs at a minimum of 3x the pixel clock so 288Mhz and the max it can handle is around 180Mhz or 192 Mhz if you are lucky.

There may be at 8 bit shift register internally outputting the video and if you could pick up the 8 bits then something might be possible as that would reduce the clock rate to 12 Mhz but even that would require some code changes.

Sch-LikA, 2021-11-21:
Allright, thanks a lot for having checked !

closed 2021-11-21 by Sch-LikA