hoglet67/RGBtoHDMI #252: CPLD successfully programmed: no menu after booting

bzeiss, 2021-11-17:
Hi, I have soldered my own board, so likely this is not a real issue with the project, but more a user mistake. As there is no discussion forum, I figured this would be the right place to ask for help before I spend more time trying to figure this out by myself.

So I have soldered the issue 4 12-bit board. I have checked for shorts on the cpld, no issues so far. When plugged the hat into a Pi Zero (no W, the cheap one), I see the rainbow screen and I can successfully program the CPLD. After programming and booting, the rainbow screen appears, but the menu does not appear, but the green led's go out on both the hat and the Pi Zero (with nothing plugged in). If I boot holding button 1 or two, I do get a screen with the resolution, no sync detected etc, but the screen hangs and I cannot do anything. Pressing all three buttons, I can reprogram the CPLD.

Do you have any ideas what I may have done wrong?

IanSB, 2021-11-19:
If you can get to the menu after holding buttons, select the "save log and EDID" function in the info menu and post the log.txt file from the SD card.
Alternatively: 
Do you have a USB TTL level serial adapter?
If so , you can connect it to the serial debug port on pins marked GND and TxD (Rx input of USB adapter to TxD)
Run a serial terminal program set to 115200 baud 8 data bit no parity
This will show what's going on

bzeiss, 2021-11-19:
Hi Ian,

thank you very much for taking the time. Those pointers were very helpful. In the meatime, I have soldered a second board that works perfectly. So I had something to reference it against. The serial interface was helpful for that. The USB serial interface reported the following which stuck out when I diffed the logs between the good and the bad board:

CPLD  Design: Legacy_3_BIT
CPLD Version: 0.0

This is what is reported after I have programmed the CPLD. So I started transplanting a new and fresh CPLD to the bad board and programmed it. It started to work. Then I made a third board and transplanted the CPLD from the bad board to it. Same problem. So I guess the CPLD is defective. The odd thing is that the programming seems to work well and after a very brief "verifying" / "verification" or so success is reported. So I assumed that the programming went well.

closed 2024-11-09 by bzeiss