hoglet67/RGBtoHDMI #254: Investigate audio support

IanSB, 2021-11-20:
Now that the zero 2W is available it seems a lot more viable to add audio support and that might even work on the original Pi depending on how it is implemented.

The first thing to try is generating an audio tone using sine wave data to confirm that we can at least produce audio.

If that works the next thing is to investigate getting audio into the Pi.

There are several possible solutions:

1. There is a peripheral built into the Pi which can receive PCM audio data but the GPIO pins are already in use so this would be a radical re-design

2. Use something like a Pi pico (which has on board A to D converters - but are they audio capable?) to capture and reformat the data to be passed into the Pi.
Actually getting the data in is going to be the biggest problem as there aren't really any spare GPIOs.
There is GPIO24 (formerly mux) which is currently used for FFOSD and also the GPIO driving the mode 7 LED could be sacrificed.
There are also a couple of GPIOs on the camera interface which could be used as general GPIOs but connecting to them would require a flexi cable.
The data requirement for 2 channel PCM audio is 48000 samples/sec x 16 bits x 2 channels which is 1536000bits/sec which could be bit banged on a single GPIO and sampled with the video so not impossible.

3. Use a usb audio dongle like this:
https://www.amazon.co.uk/Sabrent-External-Adapter-Windows-AU-MMSA/dp/B00IRVQ0F8
This is a very attractive option as it can just plug into the USB connector but would require a lot more software support to get working.
There is a bare metal USB stack for the Pi here:
https://github.com/rsta2/uspi
whic was derived from an early version of circle, a bare metal environment:
https://github.com/rsta2/circle
which could be implemented to run on one of the Pi zero2W's cores but it is just the stack and doesn't have an audio driver so that would require further investigation.

Sch-LikA, 2021-11-21:
I guess the i2s pins are already in use ? Something like this [Hat](https://github.com/skiselev/i2s_audio_phat) could be a good option too.

MarkieV, 2021-12-11:
To synchronize audio/video you need to use some sort of buffer, as video takes more processing time then audio, big change you need to buffer the video and this produces delays...
My scart RGB to HDMI converter (which has audio) clearly! has more delay then the PI RGBtoHDMI solution.

c0pperdragon, 2022-01-03:
I would of course very much prefer the USB approach, as it would allow me to keep my existing hardware. Switching to a Pi Zero 2 is no problem, but keeping my existing Pi Zero would be nicer.

emanuelelaface, 2024-04-05:
I return on this topic because it seems that someone experimented with bare metal and audio on HDMI on this project:

https://github.com/kumaashi/RaspberryPI/tree/master/RPIZEROW/Sample_HDMI_DMA_Audio_03

IanSB, 2024-04-06:
@emanuelelaface 

Thanks for that link, that looks really interesting and I will definitely be following this up.

IanSB, 2024-05-01:
@emanuelelaface 

Well that worked! I now have HDMI audio output and can play WAVs but that's only half the solution, I still need to work out how to get the audio in to the Pi in the first place.

emanuelelaface, 2024-05-02:
oh wow, that is amazing. For the input part there was the USB driver (https://github.com/rsta2/uspi) that probably solves the issue of having a GPIO available.
Anyway, thanks a lot for your work on this, it would be great to have audio directly on HDMI.

emanuelelaface, 2024-05-02:
@IanSB I noticed that in a list of Circle comments there is a tentative of USB audio with also the code for bridging two devices in streaming.
It is the example in this thread at the end: https://github.com/rsta2/circle/issues/222

And the implementation is the one attached here.
[soundbridge.zip](https://github.com/hoglet67/RGBtoHDMI/files/15186513/soundbridge.zip)

I do not know how helpful can be.

IanSB, 2024-05-03:
@emanuelelaface 
Well that thread indicates that the support only works on the Pi4 and getting it working on the zero etc would be difficult so I don't think USB audio is going to be viable for now and I will investigate some of the alternatives.

Anyway I have uploaded an Alpha test version with a HDMI audio test menu if you want to give it a try:
https://github.com/IanSB/RGBtoHDMI/issues/29

It works with the TVs/monitors I have tested so far but others have reported problems with some TVs so let me know if it works for you.

emanuelelaface, 2024-05-03:
Oh, that is amazing, I tested and I have the audio demo! I saw that the USB input is for Pi4, it is a pity because probably to use USB is readily available with the current RGB2HDMI board, while to use GPIOs will require a new design, but I understand that if there are GPIOs available probably can be much easier.
Anyway, thanks a lot for the work!