hoglet67/RGBtoHDMI #260: Intellivision Support

tlindner, 2021-12-07:
Would anyone be willing to work with me on Intellivision support?
Here is a link to a data sheet which describes the video signals available on the motherboard.
http://www.avoidspikes.com/dsplib/chips/ay38915.pdf
The input to this chip is a 5 pin digital signal which includes color and clock information.
I've successfully tested my rgb2hdmi with a CoCo 2. I have the Intellivision hardware ready to wire up.

IanSB, 2021-12-08:
This looks like it could work but there is one possible issue with the sync.
If there isn't a composite sync signal available at TTL level then you will have to create it from the V1 V2 & V5 signals by gating them together with a NAND gate.
e.g. use a 74LS10 3 input nand gate.
Connect the V2 signal to three inputs of one gate to invert it then connect that output to one input of another gate together with V1 & V5. This would mean that when V1=1, V2=0 and V5=1 (which is the code for sync) then the output will go low producing a composite sync signal.

The video connection only needs V1 to V4 so could be along the lines of the CGA mode connecting as follows to the P2 header:

V1 to RED3 (pin7)
V2 to GREEN3 (pin9)
V3 to BLUE3  (pin11)
V4 to GREEN2 (pin4)
SYNC to pin8
GND to pin3

Here is a starting profile which is set for NTSC timing with 6bpp capture and an RGBI CGA style 16 colour palette:
[IntellivisionV001.txt](https://github.com/hoglet67/RGBtoHDMI/files/7673039/IntellivisionV001.txt)
(It will probably need tweaking)

Put it in \Profiles\6-12_BIT_RGB on the SD card and reboot

If you get a stable image (with the wrong colours) the next thing to do is to create a custom palette so you will need to determine the RGB values for each of the 16 colours listed in the datasheet. Emulator source code is probably the best place to find such info and if you can find those details I will add the required palette to the next release

tlindner, 2021-12-08:
Thank you very much. I will be working on the cable.
In the mean time, I found these palette values in MAME starting on line 72:
https://github.com/mamedev/mame/blob/master/src/mame/drivers/intv.cpp

IanSB, 2021-12-08:
Here's a test palette to try:
[Intellivision.zip](https://github.com/hoglet67/RGBtoHDMI/files/7675710/Intellivision.zip)

Put it in /Palettes on the SD card and reboot then select it in the palettes menu and save the configuration.
If you have to create the sync signal as described above it might be worth connecting just the V5 signal directly to the sync input as it might see that as a sync signal. You would also have to change the sync setting from Composite to Inverted and reduce the H and V offsets in the geometry menu.

tlindner, 2021-12-08:
I just tried your suggestion to use V5 as the sync. I used "inverted" in the geometry. Here is a video of what I see:
https://user-images.githubusercontent.com/3808/145264218-c5183613-1511-40bf-9564-8b32ab9f943a.MOV
Since the screen is rolling, it seems its not quite right. I will be going to get the 74LS10 later today.

IanSB, 2021-12-08:
@tlindner 
Looks like it's locking horizontally but not vertically. Do you have an oscilloscope? If so could you capture the signal on V5 to see what sort of timing it has.

tlindner, 2021-12-08:
I do not have an oscilloscope.

IanSB, 2021-12-08:
I think I know why the vsync isn't working, can you try this updated kernel.img:
[SyncTest.zip](https://github.com/hoglet67/RGBtoHDMI/files/7678968/SyncTest.zip)
Unzip and overwrite kernel.img on the SD card

tlindner, 2021-12-08:
When I use that new kernelrpi.img my rpi zero does not boot. I don't see the four colored squares (just a black screen), and the menu button does not bring up menu.

When I switch back to the release kernel it goes back to normal.

IanSB, 2021-12-08:
@tlindner 
Sorry you need to update the whole image to the latest beta 47 then overwrite the kernel:
https://github.com/IanSB/RGBtoHDMI/releases

tlindner, 2021-12-08:
OK. I switched over to the beta, got everything set up and get this.
https://user-images.githubusercontent.com/3808/145288933-f1585661-ebbc-4ef2-9c2b-dc91d2736970.MOV
Rock solid sync, but it flashes to blank about four times a second.

IanSB, 2021-12-08:
Can you use the save log option in the info menu and post log.txt

IanSB, 2021-12-08:
@tlindner 
One further option to try is reducing the Max V height in the geometry menu as it may be trying to capture more lines than are in the frame.
Even if that works can you still post the log so I can check the timings.
The change I made to the kernel breaks some other profiles so it will have to be switchable in a full release.

tlindner, 2021-12-09:
Reducing the Max V height did resolve the blinking issue. Here is the Log file. and a Picture of the geometry I used to to see the whole screen.
[Log.txt.zip](https://github.com/hoglet67/RGBtoHDMI/files/7685297/Log.txt.zip)
[image]

IanSB, 2021-12-09:
The sync type should be inverted as the V5 signal is normally low and is blanking(sync) when high
Getting that setting wrong can lead to problems.
You will probably need to reduce the H and V offsets after that change

IanSB, 2021-12-09:
@tlindner 
An alternative to reducing the H offset is to change the "sync edge" to "leading" in the sampling menu which will have the same H timing as your incorrect sync
Also the Pixel H offset in the sampling menu should be adjusted to fine trim the video position

tlindner, 2021-12-09:
OK. Signal inverted and setting adjusted
[image]
.

tlindner, 2021-12-09:
I changed to "leading" in the sampling menu. and am now using these geometry settings:

[image]

IanSB, 2021-12-09:
This is an updated kernel.img and Intellivision profile:
[Update.zip](https://github.com/hoglet67/RGBtoHDMI/files/7685670/Update.zip)

Overwrite the kernel in the root and put the profile in \Profiles\6-12_BIT_RGB on the SD card and reboot
Select the new profile as the name has changed.

The kernel has a new setting in geometry: "V Sync Type: Blanking" which tells the sofware to expect composite blanking instead of composite sync (V5 is effectively composite blanking) and this is selected in the profile above. (This selects the change I made above). I have slightly tweaked the profile to increase the min h width to 320 to ensure all pixels are always captured

You might be able to tweak the Pixel H Offset in the sampling menu to get rid of that blue line on the left.

Once you are happy with the profile, post the saved one from /Saved_profiles
Also can you reset the converter then do a save log and post log.txt again so I can ensure things are running correctly after you have a final profile.

Finally could you post a few game screencaps for the gallery.
Press SW2 to screencap and PNG files are in the Captures folder.

BTW do you know anything about the PAL version of the Intellivision. I couldn't find a schematic but I did find an RGB converter schematic which indicated that 9 bit RGB and TTL sync were available so that would also work with the converter but with very different connections and profile.

tlindner, 2021-12-09:
I overwrote kernelrpi.img and loaded and selected Intellivision_60Hz and I got the following image. (Captured this time) I didn't change any of the settings.
[image]

tlindner, 2021-12-09:
I do not know anything about the PAL Intellivisions.

IanSB, 2021-12-09:
Select your original profile and change V Sync Type: Blanking in the geometry menu.

IanSB, 2021-12-09:
If that works, save the configuration and post the profile from /saved_profiles

tlindner, 2021-12-09:
Here is my saved configuration, Log file, and some screen shots.
Note: I am using Composite sync, if I change it to inverted, the screen becomes garbage.
[SavedProfile_intellisionsV001.txt.zip](https://github.com/hoglet67/RGBtoHDMI/files/7685873/SavedProfile_intellisionsV001.txt.zip)
[Log.txt.zip](https://github.com/hoglet67/RGBtoHDMI/files/7685876/Log.txt.zip)
[image]
[image]

Note, I never made the funky cable with the 74LS10. But it looks like it is unnecessary, correct?

tlindner, 2021-12-09:
I went back to Intellivision_60Hz and changed the H Offset to 96. And this looks good.

tlindner, 2021-12-09:
Further adjustment:
Max H Width to 368 gets rid of the green line on the left.

IanSB, 2021-12-09:
Use the save configuration to save your profile tweaks then post the Intellivision_60Hz.txt from your /saved_profiles when you are happy with it.

tlindner, 2021-12-09:
Just just played a bunch of games and I am happy with this attached Configuration.
Also, I tested powering the PI from Pin 13 of the AY-3-8915 and I saw no issues.
[Intellivision_60Hz.txt.zip](https://github.com/hoglet67/RGBtoHDMI/files/7686867/Intellivision_60Hz.txt.zip)

IanSB, 2021-12-09:
@tlindner 
>Also, I tested powering the PI from Pin 13 of the AY-3-8915 and I saw no issues.

The only potential problem with that is overloading the voltage regulator as the converter takes around 200ma so you should check if the regulator is getting too hot.

IanSB, 2021-12-09:
I've added all those changes to beta48:

https://github.com/IanSB/RGBtoHDMI/releases

tlindner, 2021-12-09:
The power transistors got too hot to touch when I powered the pi thru them. They were not too hot to touch when I powered the pi externally. So I decided not to power the pi internally.
I know that is not very rigorous, it's the safest bet when were talking about my only working Intellivision.

Thank you very much for all the work you do regarding this project.

IanSB, 2021-12-11:
@tlindner 

I've added the wiring details for the cable to the wiki, can you check it is correct:
https://github.com/IanSB/RGBtoHDMI/wiki/Cables

tlindner, 2021-12-11:
Double checked. All good.

You could add the pin numbers for the AY-3-8915 inside the Intellivision.
PIN 1 (Vss)
PIN 4 (V5)
PIN 5 (V4)
PIN 6 (V3)
PIN 7 (V2)
PIN 8 (V1)

tlindner, 2021-12-12:
There is a typo in the table - it is AY-3-8915 not AY-3-8195

Mu0n, 2022-02-06:
@tlindner I'm curious to make this work with my intellivision II. Do you happen to have links to its schematics, perhaps from the same source as the Intellivision 1 you have?

My very first step would be: to make it power up again first and get rid of that horrible 60 Hz mechanical vibration probably around the coil.

tlindner, 2022-02-08:
On Sun, Feb 6, 2022 at 3:53 PM Mu0n ***@***.***> wrote:

> @tlindner <https://github.com/tlindner> I'm curious to make this work
> with my intellivision II. Do you happen to have links to its schematics,
> perhaps from the same source as the Intellivision 1 you [...]
I do not have a stash of Intellivision schematics. I've put some requests
out to some of my friends.

-- 
--
tim lindner

"Proper User Policy apparently means Simon Says."