hoglet67/RGBtoHDMI #262: Commodore 128 80 column 60Hz

buzbard, 2021-12-12:
Hi,
I'm using my new RGBtoHDMI with a flat Commodore 128 in 80 column mode.
I've selected the "Commodore 128-80" Profile, "60Hz" for the Refresh and for the Sub-Profile I've selected "Cmd 128 60Hz" because the default 50Hz causes the screen to flash.
I've saved the profile. But when I reboot the RGBtoHDMI or power cycle, it always reverts back to 50Hz.

In the root of the SD card is a file named "profile_6-12_BIT_RGB.txt", I assume it's my selected profile configuration. It contains the following 2 lines:

profile=Commodore_128-80
saved_profile=0

The only way I've discovered to get around this issue is to go into the "\Profiles\6-12_BIT_RGB\Commodore_128-80" folder on the SD card and rename or delete the "Cmd_128_50Hz.txt" file.

If there's a better way, I'd like to know.
Thanks.

IanSB, 2021-12-12:
Flashing indicates that it can't find a timing match which in turn means there is likely something wrong with the 60Hz profile.
Can you post a screencap of the source summary in the info menu (press up and down buttons - file will be on the SD card)

buzbard, 2021-12-12:
Screen capture with the menu (btn2+btn3) only seems to capture screen without menu.
But here's a pic from my phone
[image]
.

IanSB, 2021-12-12:
That all looks sensible, can you also use the save log option and post log.txt from the sd card

buzbard, 2021-12-12:
Sure, here ya go
[Log.txt](https://github.com/hoglet67/RGBtoHDMI/files/7700081/Log.txt)
.

IanSB, 2021-12-13:
It looks like there is a mistake in the profile.
You need to set "Auto Switch" to "Sub-profile only" and then save configuration

buzbard, 2021-12-13:
Ok, that did the trick. I restored the "Cmd_128_50Hz.txt" file and now it stays with 60Hz, no more flashing!! Yah!

Thanks

IanSB, 2021-12-13:
Ok that will be fixed in the next stable release due shortly

closed 2022-02-13 by IanSB