hoglet67/RGBtoHDMI #264: Auto NTSC artifact color doesn't work on Apple IIe

anpage, 2021-12-22:
I have an enhanced Apple IIe and while manually enabling NTSC artifact color does work (and looks great, by the way,) leaving it on "Auto" never switches to color.

Using the default Apple II profile, launching The Oregon Trail leaves the output in monochrome mode. Is there something specific about my IIe or The Oregon Trail that breaks this feature?

IanSB, 2021-12-23:
If you have an NTSC IIe, I would expect it to work but I've only tried on my standard IIe which is an NTSC model. You could try experimenting by adjusting the Y Lo DAC setting a little in the sampling menu as that sets the threshold for the burst detection. Also if you have an oscilloscope you could look at the video signal to confirm if the colour burst signal is present.
Note that long cables may reduce the burst level to the point that it isn't getting detected properly.
You could also try using the internal pickup option detailed in the cables section.

It should also be noted that the auto option doesn't work on the PAL Apple IIe because it doesn't seem to produce the required colour burst signal when the PAL encoder is disabled. (The PAL encoder can't be left enabled because the PAL subcarrier interferes with the pixel sampling.) The PAL IIe should also work with auto switching if you use the internal connection option.

I believe the later models sold in PAL areas actually used Apple's proprietary "International NTSC" which was a form of NTSC encoding with a different subcarrier frequency and that required a special Apple monitor for colour display. I think that should work but I don't have one of those either. (These were basically NTSC models with a different clock frequency and 50Hz refresh rate.)

anpage, 2021-12-23:
I'm sure my machine is an NTSC model, so PAL shouldn't be an issue. I'll try adjusting those settings and see if it helps. My cable is fairly short ~15cm IDC->ribbon->RCA homemade one, but there might be some high resistance, so I'll double check that too. If those still don't solve the problem, I do have an scope I can use to check the color burst. For all I know, there's a fault with the computer itself.

I missed the internal pickup option. I might try that next either way. I've been planning on building a second RGBtoHDMI board just to leave off the analog board and use for digital.

anpage, 2021-12-23:
Adjusting Y Lo 2 or 3 levels lower fixed the problem. Thanks!

Is it worth adding a note about adjusting that threshold setting to the artifact color section of the wiki?

IanSB, 2021-12-23:
For now I've added a note to the troubleshooting section but it maybe that the video levels on the enhanced boards are slightly different. The values in the sampling menu have been set based on my NTSC IIe and PAL II+ but it really needs someone with a wide range of Apple variants to go through and look at the optimum settings.
If the DC level of the entire video signal is shifted on the enhanced model then perhaps it needs a separate profile for those models but it could equally well be an issue with your board where the burst level is too low.

closed 2022-12-20 by IanSB