hoglet67/RGBtoHDMI #266: Sub Profile setting not being saved in the menu

Pontiac76, 2022-01-09:
I think I've found a bug with saving the sub-profile.

I had a similar issue as in #262 with my Commodore 128.  The hint about the screen flashing because it can't find a lock took me down the line that there's frequency problem.

Here's what I did to fix my problem.
HARDWARE:
- Using Raspberry Pi 3B+
- Initially used an HDMI to DVI-D cable to hook up to an old monitor.  (4:3?  3:4?  I don't know the ratios)
- Switched to an HDMI wide screen 1920x1080 TV
- Commodore 128 NTSC
- Ultimate2+ to load the D81 image.

When I first booted this bad boy up, I started the 128 in 40-col mode and noticed the converter announcing its initial status screen, but everything was still black on the 128, which is expected, so I didn't notice that it was flashing black on black (theoretically? heh)

I then hit the 40/80 col button on the 128 and kicked the power.

The Basic 7 screen came up with the green on black text, but it was scrolling like mad.

I eventually found the Profile line and eventually switched it to the Commodore 128-80.  This fixed the Basic 7 screen.  It was rock solid and pixel perfect.

I then kicked up Petscii Robots, but found that the HDMI screen was flashing on and off at about 1hz. (Nod to #262)

I eventually found that in my case, I had to switch OFF the Auto Switch, and set the Sub-Profile to Cmd 128 60Hz.  The flashing stopped.

I saved the profile, it stated it saved to a 60hz text file, and all is good.  Or so I thought.

I restarted the HDMI board, with the 128 still running the game, and found that it came back up and was flashing again.  The startup screen said it was running at 50hz again?  I go into the menu, and sure enough, Auto Switch was still off, but Sub-Profile was back at 50hz.  I switched to 60, returned to the game, no flashing, went back into the menu, saved the profile, back to the game, restarted the HDMI board, and back to 50hz I went.

IanSB, 2022-01-12:
@Pontiac76  
>Auto Switch was still off, but Sub-Profile was back at 50hz. I switched to 60, returned to the game, no flashing, went back into the menu, saved the profile, back to the game, restarted the HDMI board, and back to 50hz I went.

Strictly speaking, this isn't a save bug as the selected Sub-Profile is never saved by design and it will revert to the first file in the list which is CMD 128 50Hz on next reset. Normally the only reason to switch off auto switch when using multiple profiles is so that you can manually select a profile to tweak and save it before re-enabling auto switch. When you get a flashing screen it means that something is wrong with the profile set because none of the available profile timings match the incoming signal. 

In your case it looks like Petscii Robots is using a mode with custom timing that doesn't match any of the sub-profiles so another sub-profile has to be created to match it. 
As it appears to work with  Cmd 128 60Hz, the most likely explanation is that the petscii robots timing is only slightly different from the actual Cmd 128 60Hz timing.
Can you post a screenshot of the Source Summary page in the info menu when running petscii robots (you can screencap that by pressing the Up and Down buttons -  file in /Captures on the SD card)

Maybe I will add a feature to save the sub-profile setting when auto switch is off in future but it would only be hiding the real problem which is a missing timing file.

Pontiac76, 2022-01-12:
I take back that the Basic 7 screen alone doesn't flash.  It does flash.  I don't remember what the exact situation was when I saw that it wasn't flashing, then loading the game it did flash.

Would the quick-fix be just remove the 50hz profile?  (Or rename it to something else that goes further on?)

[image]
[image]
[image]
[image]

[Log_Basic7_60hz.txt](https://github.com/hoglet67/RGBtoHDMI/files/7858506/Log_Basic7_60hz.txt)
[Log_Basic7_50hz.txt](https://github.com/hoglet67/RGBtoHDMI/files/7858507/Log_Basic7_50hz.txt)
[Log_Petscii_50hz.txt](https://github.com/hoglet67/RGBtoHDMI/files/7858525/Log_Petscii_50hz.txt)
[Log_Petscii_60hz.txt](https://github.com/hoglet67/RGBtoHDMI/files/7858526/Log_Petscii_60hz.txt)

Pontiac76, 2022-01-12:
You might notice a LITTLE bit of speckles on the Basic 7 screens, but, I've just discovered that the long length serial cable I'm using (It's the only one I could find) is temperamental when plugged into the back of the 128.  If I wiggle it around a bit, the image clears up and things look perfect.  I didn't notice that artifacting the other day, but it was really pronounced today.  Held my tongue right, so, that fixed that.  Unsure if that'll affect the above results, but, with or without the artifacting, I'm getting the flash at 50hz.

I'm using the serial cable for a couple of reasons.  This game has an adapter for the SNES game pad that plugs into the User port, and with the HDMI cable coming out, it's sitting a little awkward.  I want the ribbon that comes off the board to the serial plug to be solid and not risk "pulling" or "tugging" so I'm working on trying to make this thing sit in a solid case.

UNFORTUNATELY, the cable is a 10 foot cable, so I can understand if there's some noise that's introduced.  If you need me to redo this with the device plugged in directly, no problem.

IanSB, 2022-01-12:
The 60Hz ones have the correct timing for Basic 7 so I don't know why that's flashing now, did you set Auto switch back to sub-profile only and save the configuration?

If you look at the source summary, petscii robots is generating frames with 262 lines instead of 264 lines and that's why it can't match and so flashes the screen;

Here is an alternate 60Hz profile with 262 lines:
[Cmd_128_60Hz_Alt.txt](https://github.com/hoglet67/RGBtoHDMI/files/7858613/Cmd_128_60Hz_Alt.txt)

Put this in \Profiles\6-12_BIT_RGB\Commodore_128-80 on the SD card
Then delete the \Saved_Profiles folder on the SD card to remove any tweaks you may have made and reboot the Pi

If things work then you should notice the sub-profile change when you switch between normal and petscii modes

Pontiac76, 2022-01-12:
I understand that the cable isn't ideal.  I only wanted to make sure I had an alternative option in case things got a bit crowded in behind the machine.  With the discovery of the noise, I'll be looking to redesign the case so the cable remains protected.  I just need to figure out the orientation and physical support so I don't add stress to the PCB in the computer.

Regarding the file, yes, that seems to automatically take me where I am.  I would have expected the _alt text to show up, but its showing me 60hz at least.  No flashing, or anything of the sort.

I'll look over the previous screen shots and what this alt file has and see if I can find whats what.  I'm looking at all three files (Original 60hz and 50hz vs your alt) and there's a bunch of differences.  I'll just have a look at the two 60s later.

Thanks for the effort.  I appreciate it.  REALLY happy with this thing.

IanSB, 2022-01-13:
@Pontiac76 

Not sure why you can't see the Alt profile name but I'll add the alt profile to the next beta release.

I wonder if there is a similar difference with petscii robots on 50Hz machines?
Unfortunately I don't have a C128 myself which makes it difficult to check for this type of problem.

Can you post a few screencaps of Petscii robots and I'll add them to the gallery next time I do an update

Pontiac76, 2022-01-13:
Absolutely.  Although a bit of a mess, I'll throw some camera shots at you as well showing both monitors up and going.

Pontiac76, 2022-01-13:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Pontiac76, 2022-01-13:
Re-opened for visibility.  I consider this closed.

IanSB, 2022-01-13:
@Pontiac76 

Thanks

BTW You can actually get a second RGBtoHDMI with analog interface to work on the 40 column (VIC II) output if you fit the C128 variant of c0pperdragon's C64 mod but that's a really advanced mod.
Someone has actually done that:
https://www.youtube.com/watch?v=mvTGewKlmK4

Pontiac76, 2022-01-13:
Very interesting.  At this time though, my CRT is doing me just perfectly.  I love how my kids can hear the CRT whine, while I can't.  The image is crisp and clean, but, absolutely I'll be keeping my ear to the rail on that particular project.