hoglet67/RGBtoHDMI #267: VIC-20 support

c0pperdragon, 2022-01-16:
@IanSB
I have recently obtained a VIC-20 just for the purpose of tinkering with its video system. After modding it to generate s-video instead of composite, the signal looks actually just good enough to be converted to a pixel-perfect image.
Contrary to the C64, my PAL VIC-20 has a much lower pixel rate (4.43 MHz) which also directly matches the subcarrier frequency. This means the chroma signal behaves very predictable and normally seems to stabilize on a new color after about 1/2 of a pixel duration. Also the luma stabilizes quite quickly, so it should be possible in principle to perfectly digitize the image.  

Of course my intention here is to use the RGBtoHDMI even if the sampling bit resolution is quite limited. I am experimenting with an analog circuit that brings the luma and chroma signals to a known DC bias level, so the analog board can be calibrated to it. With 4 different slices of luma (feeding the signal to Y and the sync input) and 5 slices of chroma (using the R and B channels in parallel on the same signal) and a sample frequency as high as the RGBtoHDMI can possibly run, this may just be enough. One scan line probably has 284 pixels total (computed from the frequencies involved), so the sampling should use the highest possible multiple of that.

To test the concept, I would need a profile that just grabs the raw sample data in highest resolution. With the screen grabbing feature, I would then try to find a mapping that uses the inherent redundancy in the data to create a stable VIC-20 image (all on the PC for analysis).

IanSB, 2022-01-16:
@c0pperdragon 

Here's a x4 profile for the YUV analog folder:
[VIC20x4_50Hz.txt](https://github.com/hoglet67/RGBtoHDMI/files/7878010/VIC20x4_50Hz.txt)

c0pperdragon, 2022-01-16:
Thank you very much for the profile. I will see how far I can get with it during the week.

c0pperdragon, 2022-01-22:
I did some some more tests with the output signals from the VIC. As it turns out, the luma signal (at least on my VIC) has different levels for basically every single color which can not even be reliably separated.  Even worse, the chroma signal intensity also wildy varies between colors, not only its phase shift. 
I guess it could still be possible to figure out the digital color from these values with some fast ADCs and more signal processing, but it is quite hopeless to try this with the analog board.  

Maybe I can do something to improve the overall quality of just the analog signal, but I can see no direct way to quantize the colors for use with the RGBtoHDMI.

Consider my attempt as failed...

closed 2022-02-10 by c0pperdragon