hoglet67/RGBtoHDMI #268: Atari MegaSTE  60mhz NA machine

Biker1bob, 2022-01-26:
I have built one of the devices up with a 12bit board I purchased assembled - its an IanSB board.

This is the topic thread I am getting my info from https://www.atari-forum.com/viewtopic.php?t=40201

My situation is - in LOW or MEDIUM resolution - where I would use the RGB colour (AKA NOT MONO)  Green is substituted for Black - I cannot get the device to produce a BLACK - It is using the STE 50hz profile. My monitor is able to do 50hz. EVERYTHING else looks fine except where a black should be there is Green. 

I have adjust every setting in the software I can think of. I have rechecked all my wiring 10+ times.. I am at a loss. Any suggestions would be appreciated
[image]
[image]
[image]

IanSB, 2022-01-26:
It seems likely that one of the four green bits is stuck high.
Based on the brightness of the green it looks like the highest bit G3 is stuck on so maybe check your soldering around that signal or try disconnecting it entirely.
Are some other colours a bit 'off colour'? (White would still be white but colours would have some green added to them)

I don't have an STE but is there any video test program that displays the full colour test patterns for that machine.
I have a program like that for the Amiga which makes it easy to work out which bit is the problem. It produces an output like this:
[image]

Which shows the full 16 levels of red, green & blue and that makes it easy to confirm which bit is stuck.

Even if you can't find a program, if you post a screencap (Short press of SW2 - file on the SD card) as I can look at the RGB level of the black background with a paint program.

Biker1bob, 2022-01-27:
There is the colour test that the Diagnostic cartridge shows. I have tried removing the connections.. not sure that I did it one at a time.. aka just G3.. I will do a screen shot of the Diagnostic screen and post. Thanks for the response.

Biker1bob, 2022-01-27:
Ok, so I did some screen captures.. I also undid all the connections.. re-examined all the solder joints.. everything is good. There is another interesting development. - when the HDMI device is wired up and powered.. it does modify the output of the Monitor port. see the attached pics for the colour bars and the changes when output to Monitor power <> VGA on monitor.
[image]
[image]
[image]
[image]
[image]

IanSB, 2022-01-27:
@Biker1bob 
>it should not effect the output correct?

Correct. 
The green value in the above PNGs is 68 which implies that it is G2, not G3 that's the problem. (it would be double that for G3, half for G1 and a quarter for G0)

So repeat the disconnection test with G2.

How are you picking up the signals? directly from the chip pins or via the resistors that are connected to those pins. (if the latter, make sure you are on the right end of the resistors which might not always be the same on all resistors)  

If the pickup points are OK then it is possible that the G2 signal is shorted to a logic high somewhere in the converter

First disconnect the converter from the Atari (unplug the 16 way cable) and try powering it with a USB power supply (USB connector nearest the 16 way connector)
If you get a green screen then it means G2 is shorted to a logic high signal on one of the boards or possibly a faulty CPLD.

If you get a black screen then it's more likely that G2 is shorted to another input signal like the HSYNC or VSYNC inputs
The Hsync input has composite sync which is normally high during video (this is less likely as it would affect the sync)
The Vsync input normally has the MONO signal and I think the MONO signal is high when the RGB signal is in use.
(you can check that by disconnecting both G2 and MONO from the STE)
Check the CPLD board and extender board separately for any solder bridges or use a multimeter if you have one

Don't forget to unplug the USB power supply before reconnecting to the computer.

Biker1bob, 2022-01-28:
> Correct. The green value in the above PNGs is 68 which implies that it is G2, not G3 that's the problem. (it would be double that for G3, half for G1 and a quarter for G0)
> 
> So repeat the disconnection test with G2. If the output stays green but the monitor port looks normal that would confirm an issue with the converter and you shouldn't reconnect G2 until that is sorted.
> 
Just did this.. NOTHING changed. G2 did not affect anything on the HDMI OUT OR the Monitor out. 

> How are you picking up the signals? directly from the chip pins or via the resistors that are connected to those pins. (if the latter, make sure you are on the right end of the resistors which might [...]

Yes from the legs of the resistors just after the chip - aka before the resisted values. Several others have done in just this place so I am very sure I am correct.
 
> First disconnect the converter from the Atari (unplug the 16 way cable) and try powering it with a USB power supply (USB connector nearest the 16 way connector) If you get a green screen then it [...]

OK, so we have a green screen with just external USB power.. so I will start checking my soldering again.. and check for shorts with the multimeter

[image]

Biker1bob, 2022-01-28:
Think I found it.. On the clpd - it also had continuity with G2
[image]
.

IanSB, 2022-01-28:
>Think I found it.. On the clpd - it also had continuity with G2

Yes, that's almost certainly it, that's pins 35 & 36 which are +3.3V and G2 respectively

Biker1bob, 2022-01-28:
YES, that was it IanSB - BLACK screen now. Thankyou so much for pointing out the G2 issue.. it allowed me to follow your steps and start investigating with the multimeter knowing a start point! :)

closed 2022-01-28 by Biker1bob