hoglet67/RGBtoHDMI #269: Incorrect ZX Spectrum 128/+2 palette

shinedeveloper, 2022-01-29:
I tried the ZX Spectrum 128/+2 (grey) profile and found that the palette is incorrectly set to:
palette=RGBrgb_(Spectrum)
whereas it should be:
palette=RGBI_(Spectrum)

It manifests itself in the way that the bright colors are weird. Bright black, for instance, is greenish. When corrected, it works fine.

I believe that the ZX Spectrum +2A/+2B/+3 profile is also set the other way round. I don't have the analog extension board so I cannot test it, but it's set to:
palette=RGBI_(Spectrum)
whereas I believe it should be:
palette=RGBrgb_(Spectrum)

IanSB, 2022-01-31:
You are correct, not sure how that happened but it will be fixed on the next beta due soon.

shinedeveloper, 2022-02-01:
Great, thank you.

IanSB, 2022-02-12:
This is fixed in Beta53
https://github.com/IanSB/RGBtoHDMI/releases

closed 2022-02-13 by IanSB