hoglet67/RGBtoHDMI #271: Spectrum +3 profile

edeano467, 2022-02-05:
I'm only getting 8 colours instead of 16 on spectrum +3. Looks like just bright signals. Every colour should have 2 shades

 #
[image]

IanSB, 2022-02-05:
The default Spectrum profiles seem to have been messed up some time ago, possibly when I changed the profile file format.
Can you try this profile:
[Spectrum_+2A+3.txt](https://github.com/hoglet67/RGBtoHDMI/files/8008918/Spectrum_%2B2A%2B3.txt)

Overwrite the file in \Profiles\6-12_BIT_RGB_Analog on the sd card and delete the \saved_profiles folder to force this one to be used

edeano467, 2022-02-06:
Hi IanSB, 
That file fixed the problem thanks very much. Good work 😊

closed 2022-02-06 by edeano467