hoglet67/RGBtoHDMI #273: Dedicated luma-code input cable

c0pperdragon, 2022-02-10:
As discussed in https://github.com/hoglet67/RGBtoHDMI/issues/258 ,  it is a viable method to use this specific encoding to transfer all image data on the luma line alone. 
For this purpose a full-blown analog board is actually a bit of an overkill. Especially because I can guarantee tight tolerances on the signal levels and manual calibration should be unnecessary. 

I envision an adapter cable with an RCA plug on one side and a 12-way connector on the other side to plug into the regular 12-bit board. Directly soldered to the connector is a small PCB that carries just one MAX9144 comparator and a few passives. With a suitable zener diode and a few resistors, I should be able to create the necessary comparator input voltages. 
This would surely be much cheaper than an analog board and with the electronics in the cable itself it reduces the number of things to plug together.  

My question now here is how this cable could fit into the overall RGBtoHDMI infrastructure. Can this simply be treated like a regular digital signal source? Will this cause confusion with the solution that uses an actual analog board?

With no FPGAs in view for a long time, this is the only side of the equation I can actually do some work on right now. That's the reason I come up with these kind of ideas all the time.

IanSB, 2022-02-11:
That should work and it would be seen as a digital source with an additional digital profile.
I actually built something like your suggestion as my first experiment to capture the ZX81:
https://stardot.org.uk/forums/viewtopic.php?p=226513#p226513

It might even be possible to add the components on the main PCB for a fully dedicated board.

c0pperdragon, 2022-02-11:
I also throught about such a combined board just for the sake of the lume-code input. But this would actually not safe many components. Just a few pin headers and sockets. All the ICs still need to be there.
So the 12-input CPLD-board is surely the most versatile building block.

c0pperdragon, 2022-02-16:
After a bit of measurements, I must admit that creating a signal that can be reliably used without further calibration is nearly impossible. So even for this a programmable voltage reference is needed.
This makes a dedicated luma-only board looking sensible again.

closed 2022-03-27 by c0pperdragon