hoglet67/RGBtoHDMI #278: Flashing screen from PC in Hercules or MDA mode

oraclejedi, 2022-03-06:
I've seen a few thread on the RGB2HDMI flashing the screen on and off every half second or so.   I've tried the suggested fixes but so far nothing has helped.

I have a generic PC clone.   It has an NSI "Smart EGA" graphics adapter that can out CGA, EGA, MDA or Herc.   I have the monitor switch block set for MDA/Herc and Checkit confirms that it was it thinks the system has.

I followed "Nerdly Pleasures" blog on setting the system for Hercules.  If I have the RGB2HDMI menu up, I can see a stable crisp image from the PC.   But as soon as I exit the menu the whole screen flashes on and off. 

The monitor is a Samsung TW215.  It expects 1680x1050@60Hz.  The MDA/Herc is outputting at 50Hz I believe.   Auto-callibrate ran and reported no errors but the Sync LED is still flashing.

Note:  The system works great in EGA or CGA mode.  

Anyone got any pointers on how to stop the flashing?

IanSB, 2022-03-07:
Flashing like that means that the timings from the video source don't match any of the timings in the sub-profile folder.
Sometimes this is because the card is actually outputting non-standard timing which would work on a normal analog monitor but that won't work with RGBtoHDMI because it needs precise info in order to do pixel perfect sampling although in this case I didn't have MDA or Hercules cards so those profiles were based on published info which might not be correct.

Can you post screencaps of the source summary page in the info menu for each problem mode. To screencap, press the up and down buttons together and the captures will be on the SD card.

oraclejedi, 2022-03-07:
Thanks for taking a look at this.

Below should be the Source Summary.   At this point the PC image in the background is 100% stable.

[image]

oraclejedi, 2022-03-07:
I applied the settings from Nerdly Pleasures' blog.   Basically the Sync-Type and Clock Freq.  Without these the image is shaky.   I think the H-Offset had to be reduced to zero from the original setting of 12.

[image]

oraclejedi, 2022-03-07:
The main menu setting.

[image]

oraclejedi, 2022-03-07:
I don't want to declare victory prematurely, but I just set "Auto Switch" to OFF and I have a stable image, no menu.  Looks great.   Further testing needed....

IanSB, 2022-03-07:
>I just set "Auto Switch" to OFF and I have a stable image

Switching auto switch off disables the search for a matching sub-profile which will stop the flashing but it still means that the profile is not a perfect match for the source.

A profile is determined to be a match when three conditions are met:
1. Number of lines per frame match. This is 370 in the source and profile so that is good.
2. The sync types match. This is +H-V in both the source and the profile (after adjustment?) so again good.
3. The PPM error is less than the Clock tolerance (set to 4500 in the profile) and close to zero . The PPM error is 20870 so that is not correct.

The PPM error is determined by a combination of the Line Length and Clock Frequency settings so they need to be adjusted.
As you are not using an original hercules card there is some chance that both of the above values are non standard so you will need to adjust them as follows:

The line length may already be correct if there is no visible noise in the image but to confirm that you need to fill the screen with text all the way to the right hand side. Any noise will appear as alternating vertical bars of noisy and stable pixels and you need to adjust the line length until all the image is all clean (or all noisy if the sampling phase is set incorrectly)
Try adjusting the line length setting a few values either way to get a feel for the results.
Once you have set the line length correctly you need to adjust the clock frequency to reduce the PPM error close to zero. As you adjust it the PPM error is displayed at the top of the screen.

Once these settings are correct, save the config, re-enable the auto switch and save config again and it should select that profile.

oraclejedi, 2022-03-09:
Here's a grab of an Infocom adventure using most of the screen.    This using the settings from before - the clock at 16257000 and a line length of 864.   The image is rock solid but requires the profile auto-switch to be disabled to prevent flashing.

[image]

And here is a DOS utility that shows a screen full of text.   I have set scan lines on as it yields a more authentic looking output.   I am not seeing any distortion with these settings.

[image]

oraclejedi, 2022-03-09:
I also experimented with adjusting the clock and the line length.   Without much success.   But I did get down to 0 PPM with a line length of 883.   Auto-switching enabled did not cause flashing but the left most columns of text shimmered and were partially distorted.   I will keep experimenting to see if I can get a perfect image with a low or zero PPM error count.

[image]

IanSB, 2022-03-09:
@oraclejedi 

You adjust the line length for zero noise bars and the original 864 seems to be OK. Then adjust the clock to get the PPM close to zero so you will probably have to adjust the clock to 16000000 which is the correct value for a real hercules card but it might be some other value.
If you hold the up or down button the increase or decrease will accelerate.

IanSB, 2022-03-09:
@oraclejedi 

The published info on hercules was line length of 864 for graphics mode and 882 for text mode but your card might use only one value for both. When adjusting the line length around the correct value, as you increase from the correct value the bars will get more numerous and similarly when you decrease from the correct value the bars will get equally numerous so by monitoring the effect of increasing and decreasing you can determine the centre point where everything is clean. It is also very likely that the number will be an even value so odd numbers are probably wrong.

IanSB, 2022-03-27:
@oraclejedi 
Can you give the Alpha54E version of the software a try:
https://github.com/IanSB/RGBtoHDMI/issues/10
This has a separate PC_Hercules profile and has been confirmed to work on real hercules hardware.

oraclejedi, 2022-03-28:
Hi Ian,
Will do.   The PC1 is back on the shelf right now but I will pull it down and try the new code.  It will be an interesting exercise in upgrading the RGB2HDMI too which I have not tried before.
Should have some results by the weekend.
Thanks!
Graham

 
 

IanSB#10

oraclejedi, 2022-03-30:
I've applied Alpha 54E and it has helped a great deal.

Upon start up the RGB2HDMI goes to CGA mode which yields a corrupted display, but manually switching to PC MDA mode delivers a clear stable image from the NSI in Mono mode, albeit offset to the left.   When this is manually adjusted to an H offset of 12 the picture is centered, stable and appears to be fully rendered.

I am able to leave the auto-switch at sub+manual and I do not get the constant flashing.

Starting a game that uses Hercules mode results in the display shifting to the right with the right-most part of the display now cropped.  Manually adjusting the H offset back up to 32 shows the whole image.

Manually switching the RGB2HDMI to Herc mode yields a wobbly image with distortion visible as vertical bands in 4 locations.  The NSI hercules mode seems to much prefer the MDA settings to the Hercules ones.

The D1 LED flashes continuously.