hoglet67/RGBtoHDMI #279: Cannot revert to stable configuration

alex3331, 2022-03-19:
I am using the board with Sharp MZ-700 computer (RGBS, Acorn Atom configuration; 320x200 , 15,625 Hz/60  Hz). Until recently everything was working fine. 
I accidentally changed  some setting and vertical synchronization is no longer working - the screen is scrolling at ~1 Hz rate. Horizontal synchro and RGB seem to work fine.
I changed the SD card for a newest version of software, but it did not help. I do not know where this information is stored.
The RGBS output of my computer works fine when plugged in RGB to VGA adapter.

Any idea on how to reset everything or make this work?

IanSB, 2022-03-19:
@alex3331 
Looking at the MZ-700 schematic it appears to have RGB TTL output with composite sync so I assume you are using the Atom (TTL) profile, not the analog Atom profile?

Use the save log option in the info menu and post log.txt from the SD card. Also post a screencap of the source summary menu.

Do you have a link to any specs for the the MZ-700 video modes (PAL/NTSC resolution/refresh pixel clocks etc)  as it seems a suitable machine for a dedicated profile.

alex3331, 2022-03-19:
Somehow the VSYNC started working. Now I have an artifact in the leftmost pixel column - it is unstable. The rest of screen ic perfect. Still do not know what caused it.

I saved the files and they are in the attachment; However the file dated are  4/9/2016, although the software confirmed that the files are recorded. I had to rename EDID.bin to EDID.txt.
Source menu:
Pixel Clock : 7159030 Hz
CPLD Clk: 57272184 Hz(x8)
Clock error 9 ppm
Line duration 63695 hz
Lines per frame 262
Frame rate 60 Hz (59.92)
Sync type composite
Pixel aspect 1:1
Capture size 320x216  (640x432) - SCREEN is 320x200 and it is shown correctly
H&V range 14-333 x 24x239
Frame buffer 768x432 (768x432)
FB bit depth 4
Pi resolution (1920x1280) (1920x1280)
Pi Frame rate 60 Hz (59.92 Yz)
Pi overscan 0x0 (1x0)
Scaling 2.50 x 2.50

I will provide you with MZ-700 /MZ-800 video profiles sometime next week. MZ-800 has 2 modes RGB 320x200 and RGBI 640x200. I am sure all our user group will be very happy to use RGBtoHDMI

[EDID.txt](https://github.com/hoglet67/RGBtoHDMI/files/8309864/EDID.txt)

[Log.txt](https://github.com/hoglet67/RGBtoHDMI/files/8309857/Log.txt)