hoglet67/RGBtoHDMI #281: Feature request to cycle through artifacting on/off and the different phases on the Color Computer 3

qbancoffee, 2022-03-27:
I'm testing with the RGB connector on the Tandy Color Computer 3 and although there are sparklies, the image is very good.
I'll post an issue regarding the sparklies later....

Aside from an RGB out, the Color Computer 3 also has a composite video out in order to play Color Computer 1 and 2 games
that relied on artifacting. The RGB to HDMI adapter simulates the artifacting quite well but it's a bit cumbersome to cycle through them
as you first have to access the palette menu.

When in YUV mode for the Color Computer 1 & 2, one can cycle through the artifacting phases with SW3 of the RGB to HDMI board without having to access the palaette menu and this is quite convinient.

I was wondering if this could be implemented when in RGB mode with the difference being that artifacting on/off would be part of the cycle since artifact detection isn't done in RGB mode.

IanSB, 2022-03-29:
@qbancoffee 
>I'll post an issue regarding the sparklies later....

There are some new features that were added since the profile was created that might help with that but it's difficult for me to investigate as I don't have a CoCo 3.

>I was wondering if this could be implemented when in RGB mode with the difference being that artifacting on/off would be part of the cycle since artifact detection isn't done in RGB mode.

I'll have a look at that when possible

qbancoffee, 2022-03-29:
> There are some new features that were added since the profile was created that might help with that but it's difficult for me to investigate as I don't have a CoCo 3.

I played around with it a bit and it's better. So far the sparklies are only at the edge of the display.

 

> I'll have a look at that when possible

Fantastic, thanks!

IanSB, 2022-03-30:
One thing to try is to change from trailing edge to leading edge sync in the sampling menu. Due to the diode oring of the sync signals the trailing edge may be slow and noisy. You will need to adjust the H offset in geometry after making this change.

Also check that the line length is correct in the geometry menu (varying up and down will introduce columns of noise with the correct value having no columns of noise) Fill the full width of the screen with text of graphics when doing this.

qbancoffee, 2022-03-30:
o.k., I'll give it a shot and let you know. 
Regarding the diode oring, can you think of  a less laggy way of combining the signals?

closed 2022-12-20 by IanSB