hoglet67/RGBtoHDMI #282: Would this work with a NESRGB?

ryanm101, 2022-03-31:
NES RGB : http://etim.net.au/nesrgb/index.htm

I'm thinking the nes should be more/as basic as the amiga for rgb signals

IanSB, 2022-04-05:
If the rgb bits are available on the board to pick up them it should be possible but it depends on the bit depth of the RGB.
If it's 12 bits or less then that should be ok, greater than 12 bits and it could be made to work but the colours won't be an exact match.

ryanm101, 2022-04-06:
trying to get that info from the designer at the moment. my hope is that it is as this would be a very cheap way to add HDMI to a NES :)

ryanm101, 2022-04-06:
though a bit messy, i might look into https://www.eevblog.com/forum/beginners/converting-15khz-analog-rgb-to-digital/ to see if i can convert those analog signals to something this can work with if they are infact just analog signals

ryanm101, 2022-04-06:
> If the rgb bits are available on the board to pick up them it should be possible but it depends on the bit depth of the RGB. If it's 12 bits or less then that should be ok, greater than 12 bits and it could be made to work but the colours won't be an exact match.

@IanSB it's 6bits per colour + attenuation line. Will that work?

IanSB, 2022-04-06:
@ryanm101 

If that is six bits each for R, G and B (meaning 18 bits + intensity) Thats a total of 19 bits so that's not really going to be practical.
I notice there are a block of resistors between the FPGA and the BH7236AF and there are a lot of tracks going to that so it is likely a discrete DAC.

(If you meant 6 bits in total - 2 bits each RGB + intensity then that would be possible but given the above observation that seems less likely)

There is one other option but that would require the cooperation of the designer:
The wikipedia NES entry mentions that "The system has an available color palette of 48 colors and 6 grays" which is a total of 54 colours so the FPGA could be changed to switch into a mode that encoded the 54 logical colours as 54 different widely spaced RGB combinations which the RGBtoHDMI analog board could capture and bit extract the original 54 logical colours and then apply it's own 24 bit NES palette to those logical values. The end result is the same as having a direct digital connection to the scaler but only using the analog connections.

This is how c0pperdragons Commodore 64 and Atari 800 adapters work, they have a special RGBtoHDMI mode that does the above and allows bit exact extraction of 16 logical colours for the C64 and 256 logical colours for the Atari 800.
The other advantage of such a solution is that you only need to connect up the analog RGB signals and you still get bit exact pixel perfect scaling.

I think the designer of the NES board also produces an Atari 2600 board and that could be tweaked in a similar way to extract the 128 logical colours but that's only if he's interested in making such changes to the FPGA code.

ryanm101, 2022-04-07:
[image]
[image]

IanSB, 2022-04-07:
@ryanm101 
That circuit is pretty much what I had assumed from looking at the photo but it's actually 7 bits per colour as there are separate attenuation bits for each of RGB making 21 bits in total rather than 1 global attenuation bit. As mentioned you could pick up the top 4 bits each of RGB but you would only get approximate colours.

One other idea just occurred to me although that would need further info from the developer:
If you could get the exact RGB+attenuation bit values for all 54 colours it may be that they can be distinguished by picking up fewer than 12 bits, possibly a minimum of only 6 bits (which would distinguish 64 colours). This would allow the 54 logical colours to be discriminated and I could then apply the correct actual palette in RGBtoHDMI.
This has the advantage that no FPGA change would be required although it still means some fine soldering to pick up the signals.

From the description there are three possible palettes so may be at least one of those palettes has that possibility.

ryanm101, 2022-04-07:
I got a response: ```I'm working on a new version on NESRGB at the moment. I'll add this digital output port you request. Six bits of digital video data, one attenuation bit, sync, and dot clock.```

juj, 2023-07-14:
Hi, I'm curious if that update to NESRGB did happen? Do you know?

ryanm101, 2023-07-14:
My NESRGB didnt seem to work and i've not had time to figure out if i broke the video chip during the transplant

juj, 2023-07-15:
Ah, gotcha. Yeah, that is how these projects sometimes go. Thanks for the update!

c0pperdragon, 2023-07-16:
@IanSB 
I am already thinking about a LumaCode board for the NES. With 6 bits color depth and a pixel clock of 5.3 this would require a lumacode sample rate of about 16 Mhz. Sounds like no problem on the side of the RGBtoHDMI. Also there seems to be a RF modulator on the NES (I am not sure it this is true for all models) that could again be re-purposed. So no need to cut holes. The only difficult thing for the user would be to desolder the graphics chip.