hoglet67/RGBtoHDMI #283: Feature request: Luma-only analog board

c0pperdragon, 2022-04-03:
I am currently thinking of a cost-reduced version of the analog board with a very limited feature set: https://github.com/c0pperdragon/RGBtoHDMI/tree/master/kicad_luma_2bit/V1

It uses much fewer components and only a cheaper 2-channel DAC. The input is a simple RCA cable (maybe even directly soldered to the board to save a plug and a jack). It should work with the C64 (luma-code output) as well as the ZX80, ZX81, Apple II (I guess), as well as some others (Maybe even CGA composite).

Most of this is just the analog board with components removed, but the main difference is the DAC. The SPI interface is physically identical, but it will need a different data stream to be initialized. Unluckily, there is no obvious way to let the PI software know that this luma board is present. So there are bascially two possibilities to handle this:
1. Let the user select the board type in the sampling menu
2. Generate a data stream that will correctly set up any of the two DACs, without the Pi needing to know which one it targets. I don't know if this is even possible, but I will try to figure it out.

In any case, a modification to the Pi software would be necessary. As I don't know anything about the code, this would just be a feature request from my side. 
For my first attempt, I will configure the DAC with a simple Arduino program, but in the end this of course needs to be properly integrated into the RGBtoHDMI project.

Please let me know what you think.

IanSB, 2022-04-05:
>Unluckily, there is no obvious way to let the PI software know that this luma board is present. So there are bascially two possibilities to handle this

Actually it is possible to detect the DAC type using resistors on the SPI lines which are probed during bootup and this is already used as the software auto detects three different DACs:

MAX5259 (Used by the DIY boards)
M62364 (Used by pre-assembled boards)
BU2506 (Alternative used by pre-assembled boards)

The code also still supports a couple of DACs which were used on early prototypes but you have to edit the config to select them
(They are also available as a menu option but that is disabled in release builds to avoid users accidentally selecting them):

tlc5260 / tlv5260
dac084s085

However I'm not sure about supporting a 2 channel DAC in this way because it would need custom profiles for the DAC settings and wouldn't be able to share the existing profiles DAC values. 

A lower cost 4 channel DAC might be more feasible.

c0pperdragon, 2022-04-05:
It is not easy to actually find available DACs currently. The part BH2227FV is still available which is a reasonably priced, but only 2€ cheaper than the MAX5259 - so maybe not worth the hassle.

One advantage of my proposed 2 channel DAC is the fact that it has a built-in voltage reference, so there is no need to add a regulator.

IanSB, 2022-04-06:
@c0pperdragon 
Thinking about it, the profiles would mostly be compatible, enough for the ZX81 / Apple II etc as only refsync and refgyhi are needed and those are covered by the two channels.

I'll have to see if there are any remaining combinations of resistors that would be detectable but in any case the manual menu selection option already exists.

c0pperdragon, 2022-04-06:
A manual menu option would be very nice already. Autodetection is of course even better, if this is possible.

I am not sure about how to go with the voltage levels themselves. I guess all other variants probably are referenced with 3.3V and do a 256-step subdivisition of this. The MCP48FVB02 on the other hand can use an internal voltage reference (1.22V or 2.44V), and also has 256 steps. 
So to keep the rest of the Pi software and config compatible, a linear translation would be necessary as last step when sending data to the DAC. This would introduce a slight linearity error, but maybe this is of no big relevance. I also expect that the smaller possible range (2.44V) is still sufficient.

c0pperdragon, 2022-04-09:
@IanSB 
I have tried to understand the code for auto-detecting the device configuration, but I can not quite figure out how this process works in the case of a Rev5 board. It seems it uses some active component that responds to a query signal.
Maybe this can be extended to also detect yet another board, but I am not sure about this.
Because I can not predict how this may eventually work for my board (if at at all), I will go ahead and build a board that detects as the Rev3 analog board (normally using the MAX5259), but actually has another DAC that will be configured by an arduino for a start.

IanSB, 2022-04-12:
> I can not quite figure out how this process works in the case of a Rev5 board. It seems it uses some active component that responds to a query signal.

There is a 4.7K resistor between CLK and STB on Issue 5 boards

c0pperdragon, 2022-04-13:
I already guessed that there would be some resistor between CLK and STB. But with 4.7K pull-downs on both lines on the digital board, I assumed there would be something like a 1K resistor between CLK and STB to overcome this pull-down.

After some more study of the auto-detection code, I think that one more analog board variant can be supported. I can just place a 2.7K resistor going to +5V from the CLK line. This should cause a voltage of about 3V on the CLK line when not being driven  by the Pi (as a result of the 4.7K of the digital board pulling the voltage also to GND). As all other analog boards seem to just pull CLK to GND in idle state, this should be detectable without issue.

In fact, I was so positive about this auto-detection method, that I already ordered my prototype PCBs with such a pull-up possibility.

c0pperdragon, 2022-05-21:
I have decided to cancel this attempt as it is seems not be be versatile enough to be of much use. Instead a consolidated version that combines the analog board with the a CPLD into a single PCB would be the best option from my point of few. 
Unluckily of course all necessary parts are still running out everywhere so I put all my attempts in this direction on hold.

closed 2022-05-21 by c0pperdragon