hoglet67/RGBtoHDMI #285: MegaDrive / Gensis RGBtoHDMI

ryanm101, 2022-04-10:
According to: https://segaretro.org/Sega_Mega_Drive/Technical_specifications & https://segaretro.org/Sega_Mega_Drive/Palettes_and_CRAM its 9bit RGB (3bits per colour)

https://console5.com/techwiki/images//8/8b/Mega_Drive_1_-_VA3_-_Audio%2C_Video_%28Sheet_2%29.png shows the pins from the IC. Might need an ADC to convert 1st?

IanSB, 2022-04-12:
Although it has a low enough number of levels (9 bits per pixel) the output from the chip is already in analog. The existing analog board can only cope with 4 levels each for RGB and you would need an A to D that could accurately extract the 8 levels each from the RGB signals.

ryanm101, 2022-04-13:
Would ADV7401 do the trick?

ryanm101, 2022-04-13:
I've found and ordered a TVP7002 in the hope that will do the job

davedoc1984, 2022-08-14:
What about the Audio though?

closed 2022-12-20 by IanSB

c0pperdragon, 2023-12-29:
The Sega Master System has a simple 6-bit palette with 2 bits per R,G,B channel. So 4 levels each, which should be possible with the analog board and mounted extra comparator. Pixel clock is also quite moderate.
(I just looked into this to make sure that there is no need to come up with a Lumacode adapter ;-) )