hoglet67/RGBtoHDMI #291: EGA Colours wrong on some games

Croccy22, 2022-05-29:
I have no idea what is causing this, so not sure if it is anything to do with the RGBtoHDMI board.

I have an Amstrad PPC512 with an external EGA card connected up to an RGBtoHDMI board in EGA mode. If I load up Lemmings in EGA mode everything is fine and the colours are exactly as I would expect them to be.

However, if I load Magicland Dizzy in EGA mode, the colours are wrong. I don't have a real EGA monitor to test with so I can't tell if the graphics card is not outputting the correct colours or if the RGBtoHDMI board is doing something incorrectly. I never had an EGA system back in the day so not really familiar with how it works.

Image: [https://ibb.co/phfFYJq](url)

Any ideas what is going on here?

Thanks, Matt.

Croccy22, 2022-05-31:
Looks like the RGBtoHDMI side of things is fine then. Very difficult to find any information online as to if anyone else had issues with the game as not many people were online back then. Might have more lunch trawling some vintage BBSs lol.

From what I have read a lot of games used the Standard EGA palette for compatibility reasons, where it did give the opt to swap out the colours and use a custom palette. So I wonder if Lemmings uses the standard palette which is why it works fine and then this uses a custom palette. Who knows.

Really want to try and get this working, but at least I can know rule out the RGB board, I came across some BIOS updates for my graphics card so I may give that a go next.

Thanks for testing it out for me. Dizzy looks so good in EGA :)

closed 2022-05-31 by Croccy22

IanSB, 2022-05-31:
@Croccy22 
>I never had an EGA system back in the day so not really familiar with how it works.

EGA has two basic modes of operation:

Mode 1 is effectively CGA timing (i.e. 15Khz / 60Hz) with all 16 RGBI colours available on screen simultaneously and should work with a CGA monitor.

Mode 2 is has higher horizontal timing at 22Khz vs 15KHz, so can have 366 lines per frame (350 active) at 60Hz refresh. The mode 2 timing varies slightly between 40 column and 80 column. In mode 2 you have a palette of 64 colours using 2 bits each for RGB.

The above game uses mode 1 so should be viewable on a CGA / 15Khz monitor

>with an external EGA card connected up to an RGBtoHDMI board 

What EGA card are you using?

IanSB, 2022-05-31:
@Croccy22 
>If I load up Lemmings in EGA mode everything is fine and the colours are exactly as I would expect them to be.

Lemmings uses 80 column mode 2 (640x350) so that's a significant difference between the two games:

[image]

Croccy22, 2022-05-31:
The EGA card has a chip on it with CITYGATE D10 (https://www.vogons.org/viewtopic.php?f=46&t=76185&start=20)

Can't find much else out about it.

Interestingly, the title screen for Lemmings doesn't display properly for me either, but the main game is fine. The title screen colours are not like your screenshot and the screen is flashing so I assume it isn't matching a sub profile on the RGBtoHDMI board.

IanSB, 2022-05-31:
@Croccy22 
>The EGA card has a chip on it with CITYGATE D10

Those kinds of universal switchable cards are essentially "jack of all trades and master of none" and they tend to have compatibility issues and also produce non-standard outputs. 

Looking at the pcb it has a 16 Mhz crystal and EGA 350 line mode uses a 16.257Mhz crystal so it's timing will be out which is why you get flashing. Sometimes as well as using a non standard frequency they change the number of clock cycles per line to compensate and that would have to be tweaked as well. Their output would work on an analog EGA monitor but RGBtoHDMI needs exact timings to work so things have to be fine tuned to get a usable image

In CGA timing mode 1 it may be outputting EGA style mode 2 RrGgBb instead of RGBI (I on secondary green) and again that could be tweaked by selecting a different palette or creating a custom one so it should be possible to make a custom profile that works with that board.

Croccy22, 2022-05-31:
So if Lemmings main game works fine and the colours are all correct in that game, why would the colours be wrong in Magicland Dizzy if it's using the same display mode?

I did have to tweak the timings originally for the current modes to works without flashing, but once I had done that (I followed the instructions in one of the other issues about the screen flashing) it all seemed to work fine. I have also tried Stunt car racer, Monkey Island, Prince of Persia and Rick Dangerous. All of which work fine and display the correct colours in EGA mode.

The Dizzy games have so far been the only ones to cause me issues :(

IanSB, 2022-05-31:
@Croccy22 

>So if Lemmings main game works fine and the colours are all correct in that game, why would the colours be wrong in Magicland Dizzy if it's using the same display mode?

As you suggest, maybe the palette remapping calls have bugs
There is some info here on palette calls:
https://cosmodoc.org/topics/ega-functions/#palettes
You would need some sort of test software to investigate that.

IanSB, 2022-05-31:
@Croccy22 

Also check you are using the right game software file as there are sometimes multiple versions out there. The one I tested was 155,382 bytes

Croccy22, 2022-05-31:
I've checked my version and it is the same file size. I've also tried grabbing it from elsewhere just in case and have the same result.

The thing that puzzles me most now, Is I have been doing more reading  on the EGA standard, and apparently in 200 line mode the cards can only run with the fixed 16 colour CGA Palette. So my theory about my card not swapping the colours properly, shouldn't be the case.

I'm wondering now if it's not actually the graphics card at fault here and there is something unique in the way the game is coded. The machine I am running it on is a 086  NECv30, The game does state a minimum of a 286, so maybe they used some instructions that just weren't available on my processor. I would have expected the game to just not start or crash, but who knows.

I've also tried changing the dip switches and jumpers on my card to set it into CGA emulation mode and also telling it I had a CGA monitor attached, but always get exactly the same weird colours.

I don't even really care about playing this game on the PC as I can easily play it on any number of different systems I own. But it's just puzzling me as to what is actually going on!

I've actually got an Philips CM8833 MkII monitor here that I believe accepts RGBI input, so I may move my computer over to that tomorrow and see what happens on that monitor