hoglet67/RGBtoHDMI #292: Amiga 500 -colors are displayed incorrectly | black is dark red

sascha70, 2022-06-05:
Hello community,

unfortunately a bug is driving me crazy with the Amiga 500. 

Unfortunately, the colors are displayed somewhat incorrectly for me. What should be black is dark red. I used the Beta54 version from JanBeta and the version from main tree 20211224_c97f06a release. All have the same problem. The adapter is from LinuxJedi (https://github.com/LinuxJedi/AmigaRGBtoHDMI/tree/main/Amiga500CPLD).

Do you have a solution?

Thanks

sasch
[image]
a

abrugsch, 2022-06-05:
Who built it as that looks like one of the data lines isn't soldered
correctly on the CPLD. It might just need a re-flow. I doubt it's a
firmware bug as when I've built these, a slight colour shift usually means
one pin hasn't made a proper joint.

sascha70, 2022-06-05:
Thank you for your tip. It is indeed the connector of the Pi Zero! 

With a Pi 3 A+ it works wonderfully.

Greetings

Sascha

closed 2022-12-20 by IanSB