hoglet67/RGBtoHDMI #294: Different CPLD package

Divide128, 2022-06-16:
Sometime is hard buy cpld xc9572xl in small package. But is better available same cpld in vq64 (64pin) package. May be released JED files also for this cpld type ?

IanSB, 2022-07-11:
@Divide128 
>May be released JED files also for this cpld type ?

It is a lot of work to change to 64 pin package as the PCB will have to be relaid and it would result in increased support / testing requirements going forward as two sets of each firmware would have to be created and tested plus soldering the 64 pin package is more difficult for DIY assembly.
So far the CPLDs are more readily available than Pi zeros (or any other Pi models really) so it wouldn't help that much during the current chip shortages but I am monitoring the situation and if they are completely unavailable when Pi zero availability improves then I will reconsider this.

frank-f, 2022-07-11:
@Divide128 Have you tried to place the 64 pin IC on the board? The price and (slightly) better availability seem to be attractive, but when you realize how little the board space is, it's pretty obvious why the larger chip is no real alternative. You'd need to move at least SW2 out of the way, which changes the appearance of the adapter. I didn't like that idea and just bought a few of the 44 pin CPLDs. After all, they're not _that_ much more expensive.

closed 2022-12-20 by IanSB