hoglet67/RGBtoHDMI #295: can't force it to turn off the monitor when the screen is gray

vejinnovem, 2022-07-07:
Amiga 500 with PiStorm. Can't make RGB2HDMI to turn off, when RGB output is not used.
I updated the SD with the latest release, turned that option on, saved configuration, made sure that the sub-profile won't switch during reboots of the Amiga. It just gives me grey screen and nothing happens.
What may I do wrong?

IanSB, 2022-07-10:
Are you using the latest beta 54?
That has a grey screen fix:
https://github.com/IanSB/RGBtoHDMI/releases

vejinnovem, 2022-07-10:
I just checked. Wiped SD card, put the beta 54 release straight from zip.
Changed the option HDMI Grey Standby to On and saved configurations on both sub-profiles as well.
Still no luck.

IanSB, 2022-07-11:
Can you check that the setting is on when the grey screen is displayed as I recall that it can change standard (PAL/NTSC) during grey screen and that would select a different profile with it's own copy of the grey screen setting.
(From your post it looks like you have done that but just checking). 
Other info that would be useful:
PAL or NTSC machine?
c0pperdragon style board or linuxjedi (CPLD) style board?
Post a screencap of the source summary menu when the grey screen is on
(press up & down to screencap in a menu - file on SD card)