hoglet67/RGBtoHDMI #296: Color Compute 3 captures are the wrong aspect ratio.

tlindner, 2022-07-20:
The HDMI video produced by the device is perfect. But the screen shots it can save to PNG files are the wrong aspect ratio. Can the device be corrected to produce more accurate PNG files?
Here is how a CoCo 3 screen shot currently is saved:
[image]
This is what is produced by the CoCo 1 and 2 module that is more accurate.
[image]

IanSB, 2022-07-22:
At the moment, the screencap function just dumps the framebuffer with some minimal integer scaling if appropriate. This works reasonably well when the video output is integer scaled but can result in bad aspect ratios when using interpolated scaling because the interpolated scaling is done by the Pi's hardware scaler.
You will usually get good results with a 1920x1080p monitor because virtually all profiles will integer scale with that but lower res monitors (or forcing interpolated scaling) will lead to poorer results.

What is the resolution of your monitor?

tlindner, 2022-07-22:
That explains it.

RCA DETG160R 16" 1366x768.

IanSB, 2022-07-22:
Depending on when it was made, your monitor might be able to accept 1920x1080 and scale that down so you could try changing the resolution setting in the main menu to that. If you don't get any display, hold the menu button down during reset / bootup to reset the resolution back to Auto.

closed 2022-12-20 by IanSB