hoglet67/RGBtoHDMI #298: Elfmania game - top and bottom of the picture cut off

mvbhz, 2022-08-18:
I have a problem with Elfmania which uses overscan. On the left and right, the image is OK, but some pixels are cut off at the top and bottom. How to set it so that it does not cut?

The TV definitely displays the entire screen area. Anyway, the image is also clipped on the grabber.

I have A500, standard DENISE chip, adapter and RP0.

IanSB, 2022-08-23:
Try adjusting the vertical minimum and vertical offset settings in the geometry menu

mvbhz, 2022-08-23:
I still have to try. The last attempt led me to the fact that the image was fully visible, but it got small in the center of the screen.

IanSB, 2022-08-23:
@mvbhz 
>The last attempt led me to the fact that the image was fully visible, but it got small in the center of the screen

That means your increased capture resolution won't integer scale to the resolution of your monitor so you would have to change the scaling to interpolated scaling in the main menu to get the image to fill the screen.
What was the resolution of your monitor? A 1080p monitor would only be able to display 270 or 540 (interlaced) lines with integer scaling (270*4 or 540*2 = 1080) but if the overscan was greater than that (max would be 288 or 576 with PAL then you would have to use interpolated scaling to get those extra lines to fit.

mvbhz, 2022-08-24:
My TV has a maximum resolution of 1920x1080.

closed 2022-12-20 by IanSB