hoglet67/RGBtoHDMI #301: EGA Enhanced Mode

pierre32-au, 2022-10-29:
Following my chaotic issue #300 which I closed, this is a much more focused post. I'm running an ATI EGA Wonder 800+. I'm wondering if it would be possible to update or add a sub-profile to support the Enhanced mode of the card? Source summaries and logs for each mode attached.

EGA Mode (C80) - works perfectly:
[image]

Enhanced EGA Mode (CE80). When I switch to this in the ATI driver, the RGHtoHDMI switches the sub-profile to Mode 2, but with the flashing and "shimmering" reported in #300:
[image]

ATI mode selection:
[image]

[EGA Log.txt](https://github.com/hoglet67/RGBtoHDMI/files/9894697/EGA.Log.txt)

[EGA Enhanced Log.txt](https://github.com/hoglet67/RGBtoHDMI/files/9894699/EGA.Enhanced.Log.txt)

IanSB, 2022-10-31:
@pierre32-au 

It should be possible to support your card with appropriate profiles but further info may be needed to create such profiles.

The main issue I found with the later "enhanced" PC cards is that as well as having extra modes that need new profiles, their standard modes often used non-standard timings which is the case with the ET3000 and OTIVGA and it looks like that is also an issue with your card as the C80 mode has quite a large PPM error (normally should be less than a couple of thousand) which means that either the clock is wrong or the number of clock cycles per line (line length) is wrong.

Standard PC EGA uses a 16.257Mhz crystal and such a crystal is not present on the photos of the ATI EGA Wonder 800+ (nor a multiple of that frequency) so it is likely using an approximate frequency derived from it's crystals and that is probably the same for the above C80 mode hence the large 17532 PPM error.
Your EGA mode is also 366 lines instead of the standard 365 and the line timing is 46081ns instead of the standard  45765ns so again the clock is likely non standard and there may also be a non standard number of cycles per line

Here is a profile for mode1 and mode2 with adjusted clocks so the PPM error should be small and the correct profile should be selected with no flashing when switching between C80 and CE80:

[PC_ATI_EGA_test.zip](https://github.com/hoglet67/RGBtoHDMI/files/9903866/PC_ATI_EGA_test.zip)

Unzip and put the PC_ATI_EGA800 folder in the \profiles\6-12_BIT_RGB folder on the SD card then reboot the Pi
However it is very likely that the cycles per line value (line length) may have been changed as well and that will need adjustment.

Switch to C80 mode and check PPM error less than 4500 and no flashing 

If that looks OK fill the PC screen width with a few lines of text at the bottom and check for any columns of noise across the screen (alternating bands of noisy and clean pixels). You can also try running an auto calibration which might show the effect.

If there are columns of noise, adjust the "Line Length" setting in the geometry menu either way until they are all clean or all noisy not alternating columns (note will likely be an even number). Then adjust the "Clock Frequency" until the PPM error shown at the top of screen is close to zero. Re-run the auto calibration and finally use the save configuration option before exiting the menu.

Don't exit the menu in the middle of the above changes as that might lose you new settings before you can save them.

Then change to CE80 with no menu on screen and repeat the above.

We can look at adding any extra modes once these are working.

pierre32-au, 2022-11-04:
Mate, thank you so much for this detailed help. Dropping in those profiles and following your instructions, I have two happily operating modes that auto switch successfully when I switch them with the ATI utility.

The only flaw is this _tiny_ bit of column noise in CE80 mode which is as fine as I could tune it. See the "D" in "DRIVERS": https://youtu.be/QYMixdWt32U

If we went no further I'd be happy with that.

I suspect things will get interesting again when I start testing software to run in this mode. So far I've only tested Last Half of Darkness which runs in 16 colour VGA mode. I'm not even sure if it _should_ run on this card - but it does, with column noise that can't be tuned out with the Line Length setting. I'd like to test a wider range of things before putting together a report though.

[Log.txt](https://github.com/hoglet67/RGBtoHDMI/files/9936753/Log.txt)

IanSB, 2022-11-05:
@pierre32-au 
>The only flaw is this tiny bit of column noise in CE80 mode which is as fine as I could tune it

One or two noise columns indicates that the line length setting is off by a small amount so try adjusting the value a little in either direction (Then adjust the clock to compensate the PPM value). Make sure that text characters are all across the screen width when adjusting, typing all "M" or "W" is a good test.
If all the text goes noisy instead of all clean with just a little adjustment that is likely the right value and you get rid of the noise afterwards with the auto calibration.

>Darkness which runs in 16 colour VGA mode. I'm not even sure if it should run on this card - but it does, with column noise that can't be tuned out with the Line Length setting

I have another card that runs a 16 colour VGA mode on the EGA video output so that is possible but any other modes will require new profiles and timing info so you will have make screencaps of the source summary page in those modes.

Please post your saved profiles so far which will be in \saved_profiles on the SD card.

pierre32-au, 2022-11-07:
Yep, I got that last wrinkle ironed out with another pass at the process (and a screen full of W's). Saved profiles attached.

I haven't had time to try out any more software yet.

[Default_1.txt](https://github.com/hoglet67/RGBtoHDMI/files/9949329/Default_1.txt)
[EGA_(Mode1).txt](https://github.com/hoglet67/RGBtoHDMI/files/9949330/EGA_.Mode1.txt)
[EGA_(Mode2-40).txt](https://github.com/hoglet67/RGBtoHDMI/files/9949331/EGA_.Mode2-40.txt)

IanSB, 2023-03-04:
@pierre32-au 

Another user reported that your 800+ profile doesn't work with his 800+ card.
It's possible that these non standard modes varied with different ROM versions so what ROM version is on your card?
See here:
https://github.com/hoglet67/RGBtoHDMI/commit/8f22fdd33392456542c8bea1ca652f0f11f61ea9#commitcomment-102970201

pierre32-au, 2023-03-05:
It seems that we have the same ROM version, N 1.00. If there is any more information I can provide let me know!

[image]

IanSB, 2023-03-05:
@pierre32-au 
Thanks for replying. Your board does seem to match the other one identically, even the ATI chip markings are the same apart from a few weeks difference in production date.

Can you confirm that the EGA_Wonder_800+ profile in the latest stable release works for you:
https://github.com/hoglet67/RGBtoHDMI/releases/tag/20221221_156e2f0

This uses the files you posted above. If it doesn't work can you post the correct files that you are using.

IanSB, 2023-03-05:
What type of PC board (286, 386, 486 etc) are you using and when was it made?

One other possibility just occurred to me:
This board is capable of producing VGA style timings and in fact there is place for a VGA connector so maybe the other user's BIOS is putting it into VGA mode rather than EGA mode.

IanSB, 2023-03-05:
@pierre32-au 

I just noticed in your initial post there is an ATI config program and different settings there seem to be the most likely explanation of the differences
Is that a video bios menu or a separate config program?

pierre32-au, 2023-03-05:
I'll have to wait until next weekend to do some testing and get detailed board info as I'm away for work. The system is a Wyse Decision 386sx/16s, and I need to get in there anyway to provide some more details for this page:

https://theretroweb.com/motherboards/s/wyse-wyse-decision-386sx-16s

In the meantime here is that utility, it's from the original EGA Wonder driver/utility pack:

http://vogonsdrivers.com/getfile.php?fileid=1957&menustate=1,0

IanSB, 2023-03-05:
@pierre32-au 
I don't think you will need to check after all as it looks like your files are correct for the source summary pages you posted above but the other user has the same CGA results but slightly different timings for EGA:
Yours is 366 lines at 60hz and his is 398 lines at 55Hz but everything else is the same so the most likely explanation is a different configuration setting made by that utility.

Looking at that utility link, there are multiple options for setting the timing so really there should be profiles for all of them.