hoglet67/RGBtoHDMI #303: Selling assembled boards

lazaj, 2022-11-30:
Hello,

It's not an issue, but it can be interesting for others, so I post it here.

I have build some 12 bit boards, I saw that some people sell them (link on wiki) but senders are not in my country.
I want to know if I can sell these boards on my Tindie shop ?

Best regards.

64rulez, 2022-12-01:
Hi, I've tried to buy 2 of them today (one TTL bundle and one Atari/Amiga bundle) from the link on the wiki (SellMyRetro) but, unfortunately, the order was cancelled as I did not went to check my mail sufficiently fast. The seller don't show any more on SellMyRetro (and I don't know/understand why - it's my first use of this place).

Definitely, I'm interested .... At least, for a pair of 12 bits boards and, if possible, TTL board.

Regards - Hervé

IanSB, 2022-12-01:
@64rulez 
>I've tried to buy 2 of them today (one TTL bundle and one Atari/Amiga bundle) from the link on the wiki (SellMyRetro) but, unfortunately, the order was (automatically ?) cancelled as I did not went to check my mail sufficiently fast. The seller don't show any more on SellMyRetro (and I don't know/understand why - it's my first use of this place).

I don't know why your order was cancelled but all items are showing as available when I checked just now.

lazaj, 2022-12-02:
> @lazaj
> 
> > but senders are not in my country.
> 
> I think the current sellers I'm working with will ship to most countries.
> 
> > I want to know if I can sell these boards on my Tindie shop ?
> 
> I'd prefer that you didn't. I've spent a lot of time and money to make the pre assembled boards available including redesigning for ease of manufacture and getting them assembled in quantity. A portion of the proceeds of their sale helps to fund further development.

I know that they ship to France but from Great Britain but since Brexit shipping to EU from GB is fu**ng expensive and taxes amounts are random. A lot of retrocomputers fan in EU doesn't buy goods from GB anymore because of Brexit.
This is why I want to sell my boards.
I'am a senior electronic et embedded software engineer, my board are tested and are premium quality.

So I change my question :

Am I breaking the license by selling my cards on Tindie (regarding license files I'm shure I don't ?

lazaj, 2022-12-02:
> Hi,
> 
> Your post comes just in time !
> 
> I've tried to buy 2 of them today (one TTL bundle and one Atari/Amiga bundle) from the link on the wiki (SellMyRetro) but, unfortunately, the order was (automatically ?) cancelled as I did not went to check my mail sufficiently fast. The seller don't show any more on SellMyRetro (and I don't know/understand why - it's my first use of this place).
> 
> Definitely, I'm interested .... At least, for a pair of 12 bits boards and, if possible, TTL board.
> 
> Regards - Hervé

@64rulez I will let you know where to buy it

IanSB, 2022-12-04:
@lazaj 
>I know that they ship to France but from Great Britain since Brexit shipping to EU from GB is fu**ng expensive and taxes amounts are random. A lot of retrocomputers fans in EU doesn't buy goods from GB anymore because of Brexit.

That was certainly the case in the first 6 months or so but I understand the EU implemented the same VAT rules for selling platforms as the UK in the middle of last year so now when you buy something from a major selling platform like ebay, the VAT rate of the buyer's country is included in the final selling price and there shouldn't be any further tax charges or large handling fees payable on delivery:
https://pages.ebay.com/vat-eu/

If you only want to sell a few extra boards you have already built I guess that doesn't matter so much but if you want to do it on an ongoing basis then you may harm the project because it has progressed way beyond it's original purpose and become very expensive to maintain and add support (buying additional retro computers, getting prototypes of new interfaces made etc.) and the donations I get from the supported sellers help to pay for that.

BTW the boards from those sellers are actually improved designs with additional features such as protection components and increased analog bandwidth.

64rulez, 2022-12-10:
... As an update, I've finally managed to buy my pair of boards from the official UK seller (that had banned me by mistake in my first attempt). As of now : 
- I've been able to buy "official" boards at a "normal" price
- I have no idea about VAT at arrival (bad, but it's the game)
- So far, Royal Mail tracking says that they've got the packet and that it's supposed to go to France (nice) but no more update since one week. May be strikes or whatever, I just cross fingers. Getting items from UK was (and is still) always a kind of russian roulette

closed 2022-12-20 by IanSB