hoglet67/RGBtoHDMI #307: Amiga black screen

Bennythepipe, 2022-12-24:
I've installed all components and software according to instructions.

All I get when I power on the amiga is a black screen, nothing more.

I've tried it on three different displays.
Once on my big 65" Sony tv I got some distorted graphics on top of the screen.

Can anyone guesss what's going on ?

best regards
Niklas

IanSB, 2022-12-27:
What board are you using? e.g: c0pperdragon internal, linuxjedi internal or the standard external with internal pickup only
Did you build it yourself or was it ready built?

mikezt, 2023-05-25:
Hi!

I have had the same problem. I bought the board on EBay from retro_tinkering and after copying newest firmware to SD card, I got only black screen and the content of the SD card was corrupted. I connected serial console (log attached) and got few CMD17 errors reading SD card. So I changed the nonworking SD card (Intenso 4GB) with working 8GB noname one and the device is working fine now. I hope this can help soneone.

[serial-log.txt](https://github.com/hoglet67/RGBtoHDMI/files/11568553/serial-log.txt)

IanSB, 2023-05-25:
@mikezt 

Yes definitely something wrong with that SD card although not sure if it's just a corrupted filesystem which can be fixed by reformatting or an actual hardware fault with the card itself.
BTW the latest release has a feature which might help diagnosing issues like this because if you disconnect the Pi from any CPLD board and power it up it will display the main menu with "CPLD Not Fitted" which should show that the software is installed and booting up. (Previously doing that produced no output at all)