hoglet67/RGBtoHDMI #308: No issue. PC CHIPS G3101 / Paradise PEGA1A EGA card (AUTO MODE HEGA) profiles.

grxt, 2023-01-04:
Hello, I share here the profiles of the EGA card "AUTO MODE HEGA". It is a card with the PC CHIPS G3101 chip.

https://www.vgamuseum.info/index.php/cpu/item/476-pc-chips-g3101-paradise-pega1a

It is possible that they can be used in the Amstrad PC1640 and other Paradise PEGA1A chip cards, since it appears to be the same chip with a different name. In fact, I've used the Amstrad PC1640's DISPLAY.COM utility (once patched so it doesn't look for the "AMSTRAD" string in the BIOS) to configure the various monitors/modes on my card.

There are many profiles for the Hercules graphics mode, as I have tested several programs with native support, as well as several CGA emulators.

Sorry if this is not the right place to put it.

[PC_HEGA_AUTO.zip](https://github.com/hoglet67/RGBtoHDMI/files/10347990/PC_HEGA_AUTO.zip)

IanSB, 2023-01-06:
@grxt
Thanks, I'll add it to a future release
What software is generating so many different Hercules timings?

grxt, 2023-01-06:
Ok, thanks!

I have used native Hercules graphics modes with: Windows 3.0, Psion Chess, Korean Dungeon Boy, GhostBusters II, MiniPutt, ...

In the CGA emulation modes (which is where so many timings are generated), I have used among others the emulators that can be found on the page https://www.vogons.org/viewtopic.php?f=61&t=61035&start=160