hoglet67/RGBtoHDMI #312: Need help creating a profile for a super-rare Smaky 324

Sch-LikA, 2023-03-06:
I am part of a Swiss association that tries and get old computers back to running state. We came across a super rare Swiss computer, a 68020 based Smaky 324.

The thing used to output to a Portrait monitor we unfortunately do not have. It apparently outputs a 864x1024 resolution, probably monochrone, through a Synertek SY6845EA chip.

Here is a Salae Logic Analyzer dump of the video signals (just a few lines of text over a black background) :

[Smaky 324-sal.zip](https://github.com/hoglet67/RGBtoHDMI/files/10896675/Smaky.324-sal.zip)

Would the pixel clock be manageable by rgb2hdmi ? We found by pure luck that  the Tektronix 571 profile gave a somehow not total garbage output :
[image]

But could not get much further than that. Could somebody help ? We could document the process, that might be really helpful. Thanks a lot !!

IanSB, 2023-03-06:
Post a screencap of the source summary page which shows the measured timings.
Also there is a tutorial on how to create a new profile in the wiki:
https://github.com/hoglet67/RGBtoHDMI/wiki/Tutorial-on-Adding-a-New-Profile

To estimate the pixel clock, post the frequencies of the crystal(s) fitted in the machine.

IanSB, 2023-03-06:
@Sch-LikA 
Do you know which signals the Logic Analyzer dump are meant to be?

Sch-LikA, 2023-03-07:
@IanSB thanks a lot for helping. I did not take those dumps unfortunately I have no other details. This unfortunately is not mine, I will ask for the info !

Sch-LikA, 2023-03-07:
Sorry wrong click ...

IanSB, 2023-05-08:
@Sch-LikA 

Don't know if you are still interested in pursuing this but I recently created a test pattern generator for RGBtoHDMI that can create test video sources at any arbitrary resolution and clock rate so I used that to generate the above speculated resolution and it worked although it needed some bugs fixed that were causing issues at such a high resolution.
The latest beta58 on my github will work at 3bpp at the above resolution with an appropriate profile and here is a screencap from my test pattern generator:

[image]

Sch-LikA, 2023-05-08:
Amazing thank you so much ! 

I am still 100% interested for sure, unfortunately I currently do not have access to the machine, but I will ask its owner to test it ! 

On the other hand our association was given a pair of quite interesting Datapath Frontier Terminals. They seem quite amazing as they provide graphical Tektronix 4014 emulation (and vt52), providing a viewpoint of 736x512 over a 1024x1024 framebuffer. I'll do some testing and report. I could not find much details but I'll open an other issue if I need. 

The company is still around so I hope to get some details... 

Thanks again for your amazing work !