hoglet67/RGBtoHDMI #313: Apple IIGS Cable Info

CygnusTM, 2023-03-07:
Just suggesting some small updates to the Cables page in the wiki for the Apple IIGS section. Pin 14 goes to UF3 pin 2 on a ROM 0 or ROM 1 board, but on a ROM 3 board that chip is labeled UD1. I've also created a diagram for the ROM 3 board wiring which can be added if you'd like. See link below. 

https://i.imgur.com/lazx1Im.jpg

IanSB, 2023-03-10:
Thanks for the info.
I've been thinking of designing a buffered pickup board for the Apple IIGS which would protect the VGC outputs from damage and potentially make connection easier.

There are a couple of approaches to this:

For boards with a soldered  VGC a pcb with a PLCC socket soldered underneath could be pushed over the VGC for a plug in solution.
For boards with a socketed VGC, a thin pcb that could be pushed over the socket pins on the underside and soldered to them might work.

The main problem at the moment is that I don't have any Apple IIGS hardware at all, let alone multiple variants.

CygnusTM, 2023-03-24:
That would be great. I've found that my soldering skills are not up to this particular mod. Every time I've tried to do it I end up messing up the composite and/or RGB outputs. I gave up and got a component adapter.