hoglet67/RGBtoHDMI #316: Which connector for the extension boards?

JPT77, 2023-03-13:
Hi

I just realized the analog board parts list says

Samtec SSW-106-02-T-D-RA 2x5 way 0.1" female right angle PCB Header

but the part number is 2x6? 
I bought 2x5. 
is it possible to use 2x5 or do I need alle 12 pins?

thanks

IanSB, 2023-03-14:
@JPT77 
You can use a 2x5 connector as long as you don't want to fit U7 on the underside of the analog pcb to get the full 64 colours.
(The extra 2 pins are connected to that chip)

Only a few computers need that (Tandy CoCo3, TI99/4A)

The PCB is marked on the underside to indicate which holes to use for the 2x5 connector.

JPT77, 2023-03-23:
Thanks. And I realized that the PC-Buffer-board needs only 2x5, too. the additional pins are not connected at all?

are you gonna fix the description?
If not, I will and then try to offer a pull request from my copy of the wiki.

IanSB, 2023-03-23:
@JPT77 

I've corrected the 2x5 error in the analog board wiki, it now reads 2x6 to match the part number.

The original board I designed was the 6 bit board with a 10 way header and that used a 2x5 connector. The pinout of the 10 way header matched the EGA pinout so it was easy to make an EGA cable to plug directly into that connector.

Later I designed the 8 bit board which added 2 extra inputs to the 2x5 connector to make a 2x6 connector. This was primarily to support the full 2 bits each per R,G,B with the analog board (for 64 colours instead of 27) and that required 2 extra comparators (from U7) and two extra inputs although it was possible to use them with digital sources as well.

It was still possible to fit the old 2x5 connectors to other boards like the PC one but I increased the connectors to 2x6 on those boards so the pins and sockets always matched up to avoid confusion although I left it possible to use 2x5 connectors in case people already had bought some for the old designs.

Finally I designed the 12 bit board which added the extra 4 bits to the separate P5 connector as they couldn't be added onto the existing 2x6 way connector due to space issues.
 
(All of these designs were done in a backwards compatible way so that boards of different versions could be interchanged as much as possible.)