hoglet67/RGBtoHDMI #317: No good CPLD programming

LejonD, 2023-03-13:
I got a RGBtoHDMI for the Amiga CDTV and it starts ok with the CPLD boot screen (empty CPLD).
But when i program it with the menu (Pi Zero 2) then the named file is different than the selected file.
Im trying to program the 6-12_BIT_RGB-CPLD_v94.
and im getting the 3-12_BIT-BBC_v94.

What could be the problem?

[Log.txt](https://github.com/hoglet67/RGBtoHDMI/files/10962116/Log.txt)
[image]
[image]

IanSB, 2023-03-14:
@LejonD 
There is a 12 bit bus (Q00 - Q11) that passes video data from the CPLD to the Pi and it also transfers the CPLD type number and version number.
The version number is correct for the RGB CPLD but the type number is incorrect so it mistaking the RGB CPLD for the BBC CPLD

The type numbers are:
BBC 0x0XX
RGB 0x4XX

So it is likely that Q10 on the 12 bit bus is faulty in some way as it is stuck low. This is connected as follows:
Pin 3 of the CPLD to Pin 32 of the 40 way GPIO connector so check for continuity or shorts in that area (pin 4 of the CPLD is GND so it could be shorted to that)
Also it might be an intermittent contact on the 40 way header
Finally it could be a faulty GPIO pin on the Pi itself so if everything else looks OK then try another Pi if you have one.

This will also appear as one of the video bits stuck at zero.

LejonD, 2023-03-14:
Hello IanSB,

I did an resolder of the CPLD and a full continuity check of all the pins.
I don’t measure strange shorts or disconnected lines.
Pin 3 is connected to pin 32 on the GPIO connector. Also checked that it is connected to the Pi. Looks like all is ok. Cant check the pi because its to small.
But every thing else works, but it will not program the CPLD with the RGB version.
Will see that a neighbour has a spare Pi Zero2.

Is there a way to monitor what is happening?

IanSB, 2023-03-14:
@LejonD 
It's also possible that the CPLD itself is faulty and pin 3 is dead.
Try measuring the resistance to ground of pin 32 both on the CPLD board and the Pi (after separating them). It should be in the megaohms range. If it is a low value in the tens or hundreds of ohms range that indicates a failed pin.
If you have an oscilloscope or logic probe you could check to see if there is any activity on pin 32 during bootup.

I managed to reproduce your problem by temporarily shorting pin 32 to ground with a 5.6 ohm resistor which seems to confirm my theory.  The CPLD has RGB v9.4 programmed into it but when shorted and rebooted it reads as BBC v9.4 like yours:

[image]

(The video is messed up because the software is trying to control the RGB CPLD using the BBC driver).

>Is there a way to monitor what is happening?

There is a debug serial port that outputs the log at 115200 baud 8 data bits no parity.
Connect the RX input of a USB to TTL serial adaptor to the pins on the header marked Gnd and TxD

What happens when you try to reprogram? Does it show an error or complete the programming and reboot?
As far as I can tell it is running the RGB CPLD because the version number is read back from the CPLD and that is displayed as 9.4. 
If one of the other CPLDs was programmed the version number would show as 7.9 or 9.1 and it would take a lot of wrong bits to misread that which is why I think it's a problem with reading the bit on pin 32.

This is what the log output should look like when reprogramming:

```
Read xsvf file /cpld_firmware/6-12_bit/6-12_BIT_RGB_CPLD_v94.xsvf (length = 53373)
Programming....
CPLD programming: JTAG command = 0xfe (IDCODE)
CPLD programming: JTAG command = 0xff (BYPASS)
CPLD programming: JTAG command = 0xe8 (ISPEN)
CPLD programming: JTAG command = 0xec (UNKNOWN)
CPLD programming: JTAG command = 0xed (FBULK)
CPLD programming: JTAG command = 0xf0 (CONLD)
CPLD programming: JTAG command = 0xf0 (CONLD)
CPLD programming: JTAG command = 0xff (BYPASS)
CPLD programming: JTAG command = 0xfe (IDCODE)
CPLD programming: JTAG command = 0xff (BYPASS)
CPLD programming: JTAG command = 0xe8 (ISPEN)
CPLD programming: JTAG command = 0xea (FPGM)
CPLD programming: JTAG command = 0xf0 (CONLD)
CPLD programming: JTAG command = 0xe8 (ISPEN)
CPLD programming: JTAG command = 0xe8 (ISPEN)
CPLD programming: JTAG command = 0xee (FVFY)
CPLD programming: JTAG command = 0xe8 (ISPEN)
CPLD programming: JTAG command = 0xff (BYPASS)
CPLD programming: JTAG command = 0xf0 (CONLD)
CPLD programming: JTAG command = 0xff (BYPASS)
SUCCESS - Completed XSVF execution.
Successful, rebooting in 5 secs
Successful, rebooting in 4 secs
Successful, rebooting in 3 secs
Successful, rebooting in 2 secs
Successful, rebooting in 1 secs
```